bambulab/BambuStudio #3281: GIVE PC THE OPTION TO SKIP

Kingsidorak, 2024-01-02:
Quit splitting features to give a reason for the handy app to exist. You guys are making the overall experience horrible for people who dive deep into using your products for various uses.

I should not have to reach for my phone when I already have Bambu Studio open because it takes almost an entire minute for me to open the app and get to the specific printer thats failing a print with multiple objects only because of one object that popped off the build plate, and then I have to stop the print BECAUSE IT TOOK TOO LONG TO SKIP THE OBJECT BECAUSE YOU GUYS DON'T HAVE THAT FEATURE ADDED TO THE APP I'M ALREADY USING SINCE ALL THE MODELS I WANT TO PRINT ARE LOCAL FILES

walterwongbbl, 2024-01-22:
@Kingsidorak 
Do you want to be able to operate part skipping on a PC

Kingsidorak, 2024-01-22:
> @Kingsidorak Do you want to be able to operate part skipping on a PC？

Yes please. I got really frustrated with this and having to reach for my phone every time I heard a noise coming from the printer when I had multiple parts. Almost every time a part would fail in the time I get to my phone, and start messing up another part

thebadluck, 2024-10-13:
I second the request.  Make no sense to not have this feature in the desktop,

prdoyle, 2025-01-18:
Me too. I'm not quite as angry as @Kingsidorak, but since this feature is often used when something is going wrong and time is of the essence, the easier it is to access in a hurry, the better.

kooijtjemk2, 2025-01-24:
Much wanted and needed feature!

Kingsidorak, 2025-03-25:
@foxsam21 The Home Assistant integration for Bambu now has Skip Function with selection. It's so good. Orca Slicer could probably utilize it

okpoub, 2025-09-16:
Any news on this ?

Very frustrating to have to get my phone to do that, since i'm usually managing all the process from my computer (modeling / exporting / slicing / printer management / printing)
I never use the Bambu Handy phone app.

A button should be present in the desktop (PC) BambuStudio app.

KingSidorak, 2025-09-16:
@okpoub They've added the skip function to Bambu Studio... But right now it's only for X and H series....... 

The Home Assistant addon HA-Bambu has had the option for all printers for almost a year now

okpoub, 2025-09-17:
@KingSidorak yeah, I don't use Home Assistant (and probably neither do most basic users)
If the feature is available in Bambu Handy, i don't see why it couldn't be added to Bambu Studio for P series.

Gezafej, 2025-09-26:
add it!!!

FullPace, 2026-01-29:
Any news on this? How can this essential feature still be missing? It's ridicolous.