bambulab/BambuStudio #3284: K values don’t save on Spanish translation

xtianpaiva, 2024-01-02:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.6.3

### Printer

P1P

### How to reproduce

[image]

1. Go to Devices
2. Click te "Pen" to edit PLA (do not have AMS)
3. Write any value to the "K Factor" field and save

### Actual results

Always 0,000 is shown, no new value is stored/shown, all this with the Spanish translation, switch to English, and everything works as expected.

I think this is related to: https://github.com/bambulab/BambuStudio/issues/2446#issuecomment-1712445638

### Expected results

New K value shall be stored/shown

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-01-16:
This is a BUG related to minority languages, but we have not yet found a good way to solve this problem. We will continue to try to solve it in the future.

zhimin-zeng-bambulab, 2024-01-16:
https://github.com/bambulab/BambuStudio/issues/2546

xtianpaiva, 2024-01-16:
Around 800 million people, mainly in Europe and Latin America, use a comma for decimal representation. Tkx!

amzaldua, 2024-01-23:
> This is a BUG related to minority languages, but we have not yet found a good way to solve this problem. We will continue to try to solve it in the future

😅😅😅
The most spoken native languages, ordered by the number of speakers, are Chinese (Mandarin) at around 17.7%, Spanish at approximately 5.7%, English at about 5.5%, Hindi at around 4.5%, and Arabic at approximately 4.2%. In terms of widely spoken second languages, English leads at around 20%, followed by Spanish at approximately 6%, French at about 3%, and Mandarin at around 2%.
Spanish is not precisely a minority language! 😉

zhimin-zeng-bambulab, 2024-08-06:
It has been fixed, you can have a try in v1.9.3.50.

closed 2024-08-20 by zhimin-zeng-bambulab