bambulab/BambuStudio #3285: Wrong AMS slot chosen after slicing

EhRabz, 2024-01-02:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Additional system information

12900k, 3080ti , 32gb ram, ASUS z690-AII 

### Printer

P1S

### How to reproduce

set 2 of the AMS slots to the same filament. Choose the first one in the prepare screen, slice, and then print, and the SECOND amd slot will be chosen instead of the first one.

### Actual results

it will print from the " wrong " slot

### Expected results

print from the first slot of the same filament , or what was chosen in prepare.

### Project file & Debug log uploads

[logs.zip](https://github.com/bambulab/BambuStudio/files/13814057/logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bam-winterfell, 2024-01-05:
hi @EhRabz 
The matching of materials is only related to the type and color of the materials. Not related to the material number on the prepare screen.
Wishing you a happy life

SaltWei, 2024-01-18:
This is behavior of AMS mapping. Not bug. It will find close color and filament type.
[image]

EhRabz, 2024-03-08:
> This is behavior of AMS mapping. Not bug. It will find close color and filament type. [...]

You're not understanding what I'm saying... If A1 and A2 are both eSun PLA black , and I chose [1], in prepare screen , it defaults to A2 ... it IS a bug... EVERY previous version of BBL didn't behave like this ... and LOGICALLY it makes no sense to us the second slot of the first slot is the exact same.... Since when do we count backwards?

EhRabz, 2024-03-08:
[image]

Why would this print from A2 if A1 is the same colour , material , brand , settings, and everything?

bam-winterfell, 2024-03-19:
hi@EhRabz 
I can get you . There are two slots (e.g. a1, a3) in your AMS that are made of the same filament and color. Your model used slot 1 material in prepare page, but after slicing, it matched to slot a3.
Material matching is only related to the type and color of the filament . There is no corresponding serial number relationship between the slots on the prepare page and the slots on AMS.We can make the following modification: if there are two identical filament in the AMS, the filament with the lower serial number will be prioritized. Will this meet your requirements?