bambulab/BambuStudio #3287: Profile keeps being reset.

AdrianGarside, 2024-01-03:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Choose the profile:
[image]
[image]
[image]
[image]
2. Switch to prepare tab.
3. Resync filaments
4. Switch back to device tab
BUG: It changed the profile back to the k = 0.5 one. Which means the print quality will be awful since 0.5 is very wrong for this filament.

### Actual results

Wrong k value gets used during the print.

### Expected results

The profile I select shouldn't be reverted.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-01-03:
Hi.   @AdrianGarside 
Hello, can you provide the log file of Bambus studio

AdrianGarside, 2024-01-05:
@walterwongbbl how do I do that? I don't see any option in the UI to save logs.

walterwongbbl, 2024-01-08:
@AdrianGarside 
[image]
[image]

AdrianGarside, 2024-01-10:
I deleted that PA profile to see if that would help and now I cannot select a profile at all!
[image]
[image]
[image]
But no confirm button:
[image]
After hitting close as my only option, I'm still in the original no profile state:
[image]

AdrianGarside, 2024-01-10:
And if I reset the filament and start from scratch I can't select *any* filament into that slot!
[image]

zhimin-zeng-bambulab, 2024-01-16:
Could you have a try with firmware 01.05.01, it has improved this kind of problem.

AdrianGarside, 2024-01-16:
I already have that on my P1P and P1S but this problem is specific to my X1C.

AdrianGarside, 2024-01-16:
Unless you meant 01.07.01.00? I was running that already with at least the most recent repro and I’m on 01.07.02.00 now. I’ll see if the .2 hotfix helped.

zhimin-zeng-bambulab, 2024-01-16:
[image]

https://github.com/bambulab/BambuStudio/issues/3287
Sorry, I replied to the wrong email. I will delete these comments that are irrelevant to this subject to avoid misleading others.

Thanks！

zhimin-zeng-bambulab, 2024-01-16:
[image]
[image]

@AdrianGarside  
Could you reproduce your problem again? 
1. Turn on trace for log (the turning position is as shown above, change the severity_level in BambuStudio.conf to trace)
2. You can delete all the logs in the original log folder to ensure that the log files and your operation matches
3. Record your phenomenon (which slot was modified, what was the original value, what is the newly modified value)

AdrianGarside, 2024-01-21:
I may not be able to reproduce now since I deleted the specific profile that was involved in the repro to stop it blocking me printing with correct settings.

zhimin-zeng-bambulab, 2024-02-01:
OK.  If you encounter the same problem again, you can refer to my instructions above to enable trace, collect the logs and upload them here, and we will track the problem again. Thanks!

zhimin-zeng-bambulab, 2024-02-01:
Close as unable to reproduce

closed 2024-02-01 by zhimin-zeng-bambulab

0verEngineer, 2025-08-28:
@zhimin-zeng-bambulab https://github.com/bambulab/BambuStudio/issues/3671 Here you go, easily reproducable, the issue is present since well over a year, multiple users frustrated by it