bambulab/BambuStudio #3288: Calibration at P1 broken after Firmware 1.05.00.00

threadstone83, 2024-01-03:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

Bambu Lab P1P with P1S upgrade kit

### How to reproduce

Go to Calibration, choose Flow Dynamics of Flow rate, choose the settings you want and start the Calibration.
The Calibration starts and finishes but it's not possible to click on next to get the next view where you can enter the values.

### Actual results

It's not possible to enter the values you get from the calibration. So the calibration can not be finished.
[image]

### Expected results

Normally you will se the view where you can input the values you get from the calibration pattern and the values will be saved.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13818135/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Svanto1963, 2024-01-03:
Bambu studio : 1.8.2.56
Firmware P1S : 01.05.00.00 (20240102)
AMS1 : 00.00.06.40		

Calibration does not work.

Folw Dynamics
Flow Rate 
I can't press next when printing is done.

Solomon828, 2024-01-03:
Bambu studio : 1.8.2.56
Firmware P1S : 01.05.00.00 (20240102)
AMS1 : 00.00.06.40

Calibration does not work.

Folw Dynamics
Flow Rate
I can't press next when printing is done.

SleeperXr, 2024-01-03:
Bambu studio : 1.8.2.56
Firmware P1P : 01.05.00.00 (20240102)

Calibration does not work.

Folw Dynamics
Flow Rate
I can't press next when printing is done.

MarvAhl, 2024-01-04:
Same error with P1S and also with Flow Dynamics. In Bambu Lab Studio the progress is not displayed and it does not continue.

LucasBro21, 2024-01-04:
Bambu studio : 1.8.2.56
Firmware P1S : 01.05.00.00

Calibration does not work.

Flow Dynamics
Flow Rate
I can't press next when printing is done.

Orochium, 2024-01-05:
> Slicer 1.8.2.56, P1P firmware Firmware P1P : 01.05.00.00
> 
> Flow Calibration and Flow Dynamics Calibration cannot be completed. "Next" Button cannot be clicked to move to next part of process.

Appears resolved with 01.05.01.00 Beta firmware offer

threadstone83, 2024-01-05:
I got the 01.05.01.00 officially and already tried it. Now Flow Dynamics works. Flow Rate I can test later.

threadstone83, 2024-01-05:
Flow rate still doesn't work for me.

threadstone83, 2024-01-06:
I have to take back what I said yesterday. After a complete power cycle Flow Rata Calibration works now.

threadstone83, 2024-01-10:
Problem solved with firmware update to 01.05.01.00

closed 2024-01-10 by threadstone83