bambulab/BambuStudio #3291: Calibration BUG

bobolodrone, 2024-01-03:
### Bambu Studio Version

1.8.02.56

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

Bambu lab P1P

### How to reproduce

Go to calibration 
flow rate calibration 
manual calibration 
[image]

### Actual results

can't go on with the calibration

### Expected results

calibration completed

### Project file & Debug log uploads

[tutk_log_avapi_22700.txt](https://github.com/bambulab/BambuStudio/files/13822199/tutk_log_avapi_22700.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

capsel22, 2024-01-03:
Same for me. The print completes but cannot proceed with the prompts in Studio.

Riker107, 2024-01-03:
Same for me.

MarvAhl, 2024-01-04:
Same error with P1S and also with Flow Dynamics. In Bambu Lab Studio the progress is not displayed and it does not continue.

threadstone83, 2024-01-06:
Same problem as already reported here:
https://github.com/bambulab/BambuStudio/issues/3288

merlinz01, 2024-12-16:
A firmware update fixed the problem.