bambulab/BambuStudio #3292: Ver. 1.8.2.56 "Ironing Direction" not working

ZekeMopes, 2024-01-03:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro v. 23H2 7/12/2023, OS build 22631.2861

### Additional system information

CPU: 12th Gen Intel(R) Core(TM) i9-12900HK   2.50 GHz
Memory: 64.0 GB (63.7 GB usable)

### Printer

Bambu Lab X1C

### How to reproduce

1. Enter an ironing direction setting other than 45 degrees (e.g., 0 degrees or 90 degrees). (Other ironing settings: Ironing Type: all top surfaces, all other settings: default).
2. Slice the 3D model.
3. Scroll to a top surface layer in layer slider bar.
4. Go to toolhead path slider bar and scroll back and forth to observe ironing direction.

### Actual results

The ironing direction does not conform to the ironing direction setting. It is always 45 degrees, regardless of what was entered. (I also tried this with other Ironing Type settings: Topmost surface only and All solid layers. The same issue exists with those settings.)

### Expected results

I would expect that the ironing direction should change from 45 degrees to the angle entered by the user.

### Project file & Debug log uploads

[project file - test printing.zip](https://github.com/bambulab/BambuStudio/files/13822575/project.file.-.test.printing.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-01-05:
[image]
[image]
[image]
i use your 3mf file, and changge the ironing direction, it works fine.  Are you modify the object parameter? or you just modify the global parameter?

ZekeMopes, 2024-01-05:
I've tried this on both "Global" and "Objects" settings. Neither one works. I am trying to set the ironing direction to either 0 degrees or 90 degrees.

QingZhangBambu, 2024-01-08:
I will try to reproduce it.

QingZhangBambu, 2024-01-25:
Could you try latest version and check if the ironing direction works fine?

ZekeMopes, 2024-01-25:
Yes, I have 1.8.4.51, and I am still seeing this issue. I've attached a new .3mf file where you can see that I tried setting it to 90 degrees for ironing direction (as well as 0 degrees), and I see the same issue where the final top surface is ironing at 45 degrees (see layers 15 and 105).

[printing plate0.zip](https://github.com/bambulab/BambuStudio/files/14057439/printing.plate0.zip)

QingZhangBambu, 2024-01-26:
I got it. The ironing direction does not refer to the direction of the ironing trace. It refers to the angle between the setting direction of the ironing trace and the filling trace.

QingZhangBambu, 2024-01-26:
The name of the parameter is misleading.

QingZhangBambu, 2024-01-26:
[image]