bambulab/BambuStudio #3293: Multiple angles for rectilinear infill

cdmoras, 2024-01-04:
I'm using the X1C for research purposes and I need a finer control for rectilinear infill angles.

As of today, Bambu Studio allows to specify an "Infill direction", but that's only the starting direction for the infill, then the rectilinear infill alternates between "Infill direction" and "Infill direction"+90º.

What I need to do is to specify the angles for each layer, for example, Cura has the option to introduce **multiple angles** between brackets, and each layer will be orientated according each angle, following the order and repeating it:

[image]

I prefer to use Bambu Studio and it works great with our X1C, but I need this flexibility to continue using it for my research.

Please introduce a new linear infill pattern that allow us to use **multiple angles**  (and supports 100% infill), or modify the rectilinear pattern so it can be used with many direction angles.

Thanks.

lasse-pasker, 2024-11-28:
Thank you for bringing this up.
I have the same issue. The Rectilinear and Aligned Rectilinear options alone are not sufficient for my research. For my, it would be enough if I could set different direction for each filament.
I will try this on Cura now

tafcear, 2026-09-22:
+1 — and here is the precise gap in the current code, plus the option OrcaSlicer already ships for exactly this.

**Is your feature request related to a problem? Please describe.**
Bambu Studio cannot express a per-layer infill angle schedule. In v02.08.02.61 (latest release) the only related option is `infill_rotate_step` ("Infill rotate step", `coFloat`, min 0 / max 360 / default 0, mode `comAdvanced` — `src/libslic3r/PrintConfig.cpp` L3779-3787), and it is a *fixed* increment that only takes effect for one pattern:

- `src/libslic3r/Fill/Fill.cpp` L221-223:
  `} else if (params.pattern == ipZigZag) { params.infill_rotate_step = region_config.infill_rotate_step * M_PI / 360; }`
- `src/libslic3r/Fill/Fill.cpp` L625-630:
  `if (surface_fill.params.pattern == ipZigZag) { if (f->layer_id % 2 == 0) f->angle -= step * (f->layer_id / 2); else f->angle += step * (f->layer_id / 2); }`

So it can only add +/-N deg every two layers, on Zig Zag. It cannot express an ordered, repeating angle list such as `0,45,90,135`, and it cannot hold a constant angle on selected layers — which is what this issue (and #10782) are about.

**Describe the solution you'd like**
Port OrcaSlicer's two options:
- `sparse_infill_rotate_template` — OrcaSlicer `src/libslic3r/PrintConfig.cpp` L4396-4407: comma-separated degrees applied in order per layer, repeating when the list ends; advanced metalanguage `+5` (rotate +5 deg every layer) and `+5#5` (rotate +5 deg every 5 layers); the tooltip notes "When a template is set, the standard infill direction setting is ignored."
- `solid_infill_rotate_template` — same file, L4410-4419.

Both are `coString` / `comAdvanced`. OrcaSlicer applies them per layer in `src/libslic3r/PrintObject.cpp` L3451 and registers them as per-object override keys at L1447-1448.

**Describe alternatives you've considered**
- `infill_rotate_step` is strictly weaker: fixed step, Zig Zag only, no way to pin an angle on chosen layers.
- Rotating the model itself rotates the whole part, not just the infill.

**Additional context**
UI placement request: Strength tab -> "Sparse infill" group, next to `Infill direction` (Bambu Studio `src/slic3r/GUI/Tab.cpp` L3324 in v02.08.02.61). The plain comma-separated list form would already cover the common cases; the full metalanguage would be a bonus. Porting it as-is would also close #10782.

tafcear, 2026-09-22:
One more data point for whoever ends up picking this up.

There is now an active effort in this repo to port fixes back from OrcaSlicer — #12298, #12302,
#12303, #12306 (by @fer-marino). Flagging it here because the infill rotation template is a
possible candidate for that route, but being explicit about the scope difference: those PRs are
few-line guards, whereas this is a *process setting* port. It would touch the config system (a
new `coString` option), the slicer core (per-layer infill angle), the Strength tab, and the
translation files.

The OrcaSlicer side is small and self-contained if that helps:
- `src/libslic3r/PrintConfig.cpp` L4396-4407 (`sparse_infill_rotate_template`) and L4410-4419 (`solid_infill_rotate_template`)
- applied per layer in `src/libslic3r/PrintObject.cpp` L3451, registered as per-object override keys at L1447-1448

@fer-marino — no pressure at all, and please just ignore this if your series is deliberately
scoped to crash and build guards. If a feature port like this is out of scope for the batch you
are carrying, that is a perfectly useful answer too.

fer-marino, 2026-09-22:
Thanks @tafcear, that's useful, and you're right about the difference in scope. The PRs so far are small crash and bounds guards, while the rotation template is a setting port: new options, per-layer angle handling in the slicer core, the Strength tab, and translations.

I'd like to take it on, but after the current series. Right now I'm getting the unit-test suite building and passing again, so ports like this can land with tests. It would come as its own PR, including OrcaSlicer's later fix for rotation templates on raft prints (OrcaSlicer/OrcaSlicer#14894), so the port doesn't bring that crash with it.

Thanks for the pointers to the Orca code, they'll save time.

tafcear, 2026-09-23:
That is a good catch on OrcaSlicer/OrcaSlicer#14894 — a naive port would have carried the raft crash straight over. Landing it as its own PR with tests makes sense; no rush at all.

One thing I can offer in return: I have a Bambu Lab H2D and a P1S on hand, so once your PR has a CI build I can test it on real hardware — in particular the raft + rotation template combination from #14894 (`raft_layers >= 1` with a `"+45"` template), on both machines, and report back before it goes in. Say the word if that would be useful.

fer-marino, 2026-09-24:
Opened as #12362, including the raft fix from #14894 as promised.

@tafcear, thanks again for the pointers to the Orca code — they were accurate down to the line numbers, which saved real time. The CI-build offer for the H2D/P1S raft + rotation template combination would be genuinely useful once the PR has a build; I'll ping this thread when that's up.