bambulab/BambuStudio #3294: Bambu Studio crashing when loading 3mf or switching printers

win10ha, 2024-01-04:
### Bambu Studio Version

1.8.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

CPU: I7 4770K
Ram 16GB DDR3
GPU: GTX 1070Ti

### Printer

Bambu A1

### How to reproduce

Open Bambu Studio and attempt to load a painted 3mf file, also occurs when switching printers 

### Actual results

Slicer thinks for a second an than crashes, no dialog after the crash 

### Expected results

Files should open without a crash and I should be able to Switch printers, this is also happening in Orca Slicer since the A1 was added

### Project file & Debug log uploads

[Crash.zip](https://github.com/bambulab/BambuStudio/files/13825394/Crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

win10ha, 2024-01-04:
DMP file was too large to be added

bambu123, 2024-01-05:
> DMP file was too large to be added

Can it be shared through OneDrive or Dropbox, etc.?

win10ha, 2024-01-05:
> > DMP file was too large to be added
> 
> Can it be shared through OneDrive or Dropbox, etc.?

Attempting one drive now, issue also now occurs with all 3mf files

please let me know if the dmp file contain what you need, the log folder has many in it

[image]

bambu123, 2024-01-05:
> Attempting one drive now, issue also now occurs with all 3mf files
>
> please let me know if the dmp file contain what you need, the log folder has many in it [...]

Please provide random two of them.

win10ha, 2024-01-05:
> Please provide random two of them.

Here you go, it includes 3 DMP files 

https://www.dropbox.com/scl/fi/n7rowemibzzxir1w6d5qk/DMP-Filesx4.zip?rlkey=zkd2bot3xpcum3wbdg7xxap34&dl=0

win10ha, 2024-01-05:
> Please provide random two of them.

This may help as well

[image]

bambu123, 2024-01-08:
How did you get your version? We don't have version 57, so we can't analyze DMP files.

win10ha, 2024-01-08:
> How did you get your version? We don't have version 57, so we can't analyze DMP files.

It was posted on here as a resolution for someone else having crashing issues, its internal version 57.

would you like DMP files using the current release version?

bambu123, 2024-01-08:
> > How did you get your version? We don't have version 57, so we can't analyze DMP files.
> 
> It was posted on here as a resolution for someone else having crashing issues, its internal version 57.
> 
> would you like DMP files using the current release version?

Yes, please. If the release version can also be reproduced

win10ha, 2024-01-08:
> > > How did you get your version? We don't have version 57, so we can't analyze DMP files.
> > 
> > 
> > It was posted on here as a resolution for someone else having crashing issues, its internal version 57.
> > would you like DMP files using the current release version?
> 
> I just found this version and can see DMP information now. No need for new DMP for now. But please help me check if the latest version has fixed this problem.

Here ya go, updated files with 2 crashes (1 when switching from Bambu A1 to my CR10 the other when attempting to open a 3MF on my A1)

https://www.dropbox.com/scl/fi/bw9ggp8g7zxk9hrs2galk/DMPNew.rar?rlkey=n8r372n6gwcctyd3g5r4pypul&dl=0

bambu123, 2024-01-08:
> > > > How did you get your version? We don't have version 57, so we can't analyze DMP files.
> > > 
> > > 
> > > It was posted on here as a resolution for someone else having crashing issues, its internal version 57.
> > > would you like DMP files using the current release version?
> > 
> > 
> > I just found this version and can see DMP information now. No need for new DMP for now. But please help me check if the latest version has fixed this problem.
> 
> Here ya go, updated files with 2 crashes (1 when switching from Bambu A1 to my CR10 the other when attempting to open a 3MF on my A1)
> 
> This is on the latest version, fresh install
> 
> https://www.dropbox.com/scl/fi/bw9ggp8g7zxk9hrs2galk/DMPNew.rar?rlkey=n8r372n6gwcctyd3g5r4pypul&dl=0

From the log, it seems that the parameters modified before saving crashed. Which parameters did you modify before?

win10ha, 2024-01-08:
> > > > > How did you get your version? We don't have version 57, so we can't analyze DMP files.
> > > > 
> > > > 
> > > > It was posted on here as a resolution for someone else having crashing issues, its internal version 57.
> > > > would you like DMP files using the current release version?
> > > 
> > > 
> > > I just found this version and can see DMP information now. No need for new DMP for now. But please help me check if the latest version has fixed this problem.
> > 
> > 
> > Here ya go, updated files with 2 crashes (1 when switching from Bambu A1 to my CR10 the other when attempting to open a 3MF on my A1)
> > This is on the latest version, fresh install
> > https://www.dropbox.com/scl/fi/bw9ggp8g7zxk9hrs2galk/DMPNew.rar?rlkey=n8r372n6gwcctyd3g5r4pypul&dl=0
> 
> From the log, it seems that it crashed while saving the modified parameters. Which parameters did you modify before?

This was a fresh install, I have custom profiles for my other printers. What information can I delete to start from scratch?

bambu123, 2024-01-09:
> This was a fresh install, I have custom profiles for my other printers. What information can I delete to start from scratch?

Please config to have a more detail DMP.
HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\Windows Error Reporting\LocalDumps\bambu-studio.exe
DumpType=2

win10ha, 2024-01-09:
> Please config to have a more detail DMP. HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\Windows Error Reporting\LocalDumps\bambu-studio.exe DumpType=2

Ok here are the files with the registry change and the same two crash scenarios 

https://www.dropbox.com/scl/fi/zpswyaqu7abi5ss9xtmwg/DMPtype2.zip?rlkey=irnfegry2l9cego9jpfjql0l1&dl=0

win10ha, 2024-01-17:
> Please config to have a more detail DMP. HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\Windows Error Reporting\LocalDumps\bambu-studio.exe DumpType=2

Any update?

bambu123, 2024-01-17:
Relating to head_wrap_detect_zone parameters, in the Bambu series of presets, this is an empty array.

win10ha, 2024-01-17:
> Relating to head_wrap_detect_zone parameters, in the Bambu series of presets, this is an empty array.

I'm glad I was able to find a bug, making the app experience better.