bambulab/BambuStudio #3297: Bambu Studio - A1 purges every layer even with single color

Myrddinwylit, 2024-01-04:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.1.2

### Printer

Bambu A1

### How to reproduce

1. Just slice any file that is not from makerworld (.Stl or .3MF) after you slice if you turn on the travel display you will see that the machine goes to the purge every layer of the print.
If you slice the same file using with the my X1C printer it slices fine no purge issue per layer

### Actual results

If sent to the printer it attempts to purge on each layer - sometimes partilly in the layer. the purge process is not a consistent purge - the purge fan comes on once or not at all.  the Purged filament ends up in the bed or stuck in the purge mechanism

### Expected results

Normal slice and print

### Project file & Debug log uploads

[Collar.3mf.zip](https://github.com/bambulab/BambuStudio/files/13830746/Collar.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

coreyward, 2024-01-06:
^^^ Yeah the A1 and A1 Mini move the print head to the X-axis mounted wiper when taking the photo for the timelapse, but the printer does not actually extrude during the move. Turning the timelapse off will remove this extra travel step on each layer, though.

rick-77, 2024-02-03:
i have the same problem, and the timelapse function is not the problem.  I never enabled this option and still have purges after every layer, its very annoying because the poop stucks to the printbed and causes printfails.

MadMax389, 2024-07-05:
I think it checks to see if smooth time lapse is enabled. If not, it skips the commands to travel to the edge of the plate between M622 and M623. It does throw off the time estimate in the preview.
[image]

gaolegao-lx, 2024-07-08:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

MadMax389, 2024-07-08:
@gaolegao-lx  Which user's log files are you requesting? You can see this behavior by slicing a cube primitive for the A1. Travel time is showing 17 minutes, but it doesn't actually make the travel during the print. 
[image]

When sliced for P or X1C, these travel moves are not shown.
[image]

coreyward, 2024-07-08:
@MadMax389 And the total printing time does not include that value. I don't really know what you want here—the final determination for whether or not timelapse is enabled happens after the slicing happens (whether on the machine or in the send-to-printer dialog). How is the slicer to know whether or not to show these travels and the extra time they take?

MadMax389, 2024-07-08:
@coreyward Yeah, I understand, and it's not a major problem for me. My point is that the slicer does know if you have selected traditional vs smooth time lapse. When sliced on an X or P series, the travel lines only show up when you have selected smooth time lapse, unlike the A series where they show up regardless of the time lapse selection. For X and P, the conditional "M1002 judge_flag timelapse_record _flag" command is only inserted when smooth is selected. In either case, when smooth is selected, the time estimate will be off by the amount of travel if time lapse is ticked on the send print screen.

In previous versions of Studio (1.7 maybe), you enabled the time lapse function on the slicer screen, not the send print screen. In that case, it would generate an accurate time estimate based whether time lapse was enabled or not. When they moved it to the send print screen, there is no way for the slicer to know whether it is enabled or not. Bambu really should change the A gcode to match the X/P gcode so the travel lines don't show up if smooth is not selected.

wanchooco, 2025-06-22:
this is a good discussion and still an issue (for me)...@MadMax389 agree with your mild frustration here. The issue only seems to manifest on A1s and mainly impacts (for me) to understand the travel paths the tool head is actually going to take.
...the slice time is only a slight annoyance with no real effect but would be nice to get this right. 
Wonder if there is a request option to allow turning on/off the time-lapse feature prior to slicing which should allow this to work (apologies if I come across as simplifiying) :D

Jesper-Hustad, 2025-11-10:
I also have this issue!