bambulab/BambuStudio #3299: Blobs produced at end of long straight lines on every layer when doing spiral mode

davidjdixon, 2024-01-04:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.2

### Printer

Bambu Lab X1 Carbon

### How to reproduce

In my example:
1. Use a 0.8mm nozzle, set line-height to 0.8mm, outer wall width to 2mm (this is an extreme setting, but works in the right circumstances).
2. An an object with a straight edge of around 13mm followed by a curved corner (a simple curved rectangle would be fine)
3. Set print to work in spiralised (vase) mode with zero bottom layers
4. Start print and watch results at the junction of the long straight line and the curve.

### Actual results

At each layer, the print head will pause for a fraction of a second and purge excess material at the exact same position.
This is not a traditional seam line as the print head isnt jumping to a new layer, but instead it seems to pause to finish an extrusion is believes hasnt fully completed.
[image]

Note, this only happens at the end of long straight lines, by adding a tiny curve along the long line, it stops this purging behavior.
[image]

### Expected results

The print head should not be pausing or creating such excessive extrusion.

Note: within Bambu Studio, there is no indication that this pause is occuring, and no visible seem is shown in the rendered output in any of the visualisation modes.

### Project file & Debug log uploads

n/a. common issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-01-05:
if you don't mind, could you provide the 3mf file you used?

davidjdixon, 2024-01-29:
[example-blobs.3mf.zip](https://github.com/bambulab/BambuStudio/files/14085804/example-blobs.3mf.zip)
Apologies. I didnt see the notification for this. I have attached the 3MF for review

davidjdixon, 2024-01-30:
For reference, I'm also attaching a small video that shows how the printer is behaving during the print that creates the issue (ie the pausing/purging)

https://github.com/bambulab/BambuStudio/assets/452007/1691f6dc-2b85-4956-b2a4-160568680ed2

davidjdixon, 2024-01-30:
For additional comparison, I've added a new plate to the 3MF which has the exact same model with small negative cylinders added to create a tiny "bump" in the long lines. This "hack" stops all blobbing for a reason I cant fathom.

[example-blobs.3mf.zip](https://github.com/bambulab/BambuStudio/files/14101387/example-blobs.3mf.zip)

QingZhangBambu, 2024-01-31:
I slice the 3mf file there is no abnormal E in gcode cmd. Could you provid the gcode file for us to print in machine to reproduce it?

QingZhangBambu, 2024-01-31:
[image]
And The version shows 1.9, Bambu Studio does not generate the 3mf.

QingZhangBambu, 2024-02-04:
thanks for the files. i will check this and talk to concerned staff. and if you don't mind you can post the PR link you push to orca, concerned staff will check, if they merge the code, we will thanks you in release note.

davidjdixon, 2024-02-05:
Thanks and not a problem. The link for the Orca Slicer issue has been reported here: https://github.com/SoftFever/OrcaSlicer/issues/3461

mxbng, 2024-04-11:
Hello everyone,

Is there a solution to this problem?
I have the same

QingZhangBambu, 2024-04-11:
@davidjdixon Holle, this issue has been fixed, please update to the newest firmware

LJorissen-1, 2024-04-11:
@QingZhangBambu 
Hi,

I've noticed a similar issue on my P1S, but it seems there is only a firmware update available for the X1c. Will a fix for the P1S printers follow?

Additional info related to P1S: while printing wider objects with the same settings as davidjdixon (nozzle: 0.8mm, layer-height: 0,8-1mm, line width: 1mm - I was using these settings because I'm interested in the shape of the lines on the first layer. I'm not planning to use these settings on multi-layer prints), printing is fine over short distances (e.g. about 5cm). However, at the end of longer lines (20cm) the print head stops moving and keeps extruding material for a few seconds, resulting in a huge blob at the end of the line.

I think I have a timelapse video of this on my printer. I will upload it later if necessary.

mxbng, 2024-04-11:
@LJorissen-1 @QingZhangBambu 
Hi, 
i also have a P1S. Is the update related to the printer or the slicer?

LJorissen-1, 2024-04-11:
I assume that it is an update related to the printer, considering that QingZhangBambu asks to update to the newest firmware. 

According to [this page](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history) there is a new firmware update for the X1c released on the 8th of April.

The last firmware update for the P1S was released on the 22nd of January according to [this page](https://wiki.bambulab.com/en/p1/manual/p1p-firmware-release-history).

QingZhangBambu, 2024-04-12:
@LJorissen-1 Relevant staff stated that p1 was repaired, please update and check. If you still have problems, please open a new issue and upload the 3mf and gcode files and I'll talk to the concerned staff.

MBLine, 2024-04-12:
@QingZhangBambu 
I have open a new issue #4987 
https://github.com/SoftFever/OrcaSlicer/issues/4987

QingZhangBambu, 2024-04-17:
@mxbng @LJorissen-1  OK, I‘ve talked to them and the concerned staff said that it will fixed in the next P1S firmware version.