bambulab/BambuStudio #3301: Calibration proble,

Ianjtl, 2024-01-04:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 pro

### Printer

Bambu Lab P1S

### How to reproduce

When doing a flow rate calibration with the latest firmware on the P1S, the printer does the prints all the different flow rates for the course flow rate, however you can not click next to do the fine calibration

### Actual results

Nothing, you can not set the flow rate

### Expected results

Hopefully for this to be fixed so we can calibrate the printers

### Project file & Debug log uploads

[Bambu Log.zip](https://github.com/bambulab/BambuStudio/files/13834248/Bambu.Log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

DaHoov100, 2024-01-05:
Same issue here. Just upgraded from P1P to P1S with the kit, along with firmware upgrade. Now I can't recalibrate any flow issues. Anyone have a way to do it 100% manually?

ittmax, 2024-01-05:
> Same issue here. Just upgraded from P1P to P1S with the kit, along with firmware upgrade. Now I can't recalibrate any flow issues. Anyone have a way to do it 100% manually?

Just did a firmware check on P1S and it looks like it solved it for my instance

mhellnerdev, 2024-01-06:
Same here. Hotfix released today resolved it.

threadstone83, 2024-01-06:
It's the same problem as here reported here:
https://github.com/bambulab/BambuStudio/issues/3288

walterwongbbl, 2024-01-18:
Please update the latest firmware as this issue has been resolved and fixed.

closed 2024-01-18 by walterwongbbl