bambulab/BambuStudio #3304: libwebkit error when launching ubuntu appimage

telderfts, 2024-01-04:
### Bambu Studio Version

1.08.02.56

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

BambuLab X1C

### How to reproduce

When running the appimage, even with proper permissions, it doesn't run. When trying to run it via terminal it gives the error "/tmp/.mount_BambuSIeGQ2F/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory"

### Actual results

errors out

### Expected results

should run.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-01-05:
@telderfts it is a known issue on ubuntu

currently please install this library by yourself 
`sudo apt-get install libwebkit2gtk-4.0-dev`

thanks

lanewei120, 2024-08-08:
we also provided a ubuntu 24 version which includes libwebkit
you can have a try

closed 2024-08-08 by lanewei120