bambulab/BambuStudio #3305: Multicolor print only works on certain orientations of an object

zipforth, 2024-01-04:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: ryzen 7700x
GPU: radeon 7800XT
RAM: 64GB ddr5
storage: 2TB NVME

### Printer

Bambulab A1 with AMS Lite

### How to reproduce

After coloring a print of mine using the studio's color tool, I tried printing the object (picture attached) flat, but the sliced version is missing all colors and just prints monotone. Printing with the object rotated so its thin side is on the bed gives the multicolor expected. You can see that in both screenshots, the prepare window in the top left shows that it should be colorized.
[image]
[image]
This is rather annoying as printing this model vertically has far more material wasted in color changes than if it were printed flat.

### Actual results

Actual results are a blank object in a single color in the slicer when printing flat, despite the prepare section showing color. I attempted to send it to the printer anyway to see if it's just a visual bug, but the printer options show the AMS would only be using the one color of filament (screenshot attached).
[image]

### Expected results

Prints should be colored according to the color tool no matter the orientation of the print. If orientation is an issue for quality of print, a warning should pop up allowing the user to either re-orient the print or proceed anyway.

### Project file & Debug log uploads

[bugged.zip](https://github.com/bambulab/BambuStudio/files/13836298/bugged.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-01-12:
Thanks for reporting this issue, we will further investigate the cause of this BUG

zhimin-zeng-bambulab, 2024-01-15:
[image]
This model is composed of multiple overlapping geometries. In the overlapping parts, one color is 3 and the other is 4. When two colors overlap, we display the color with the smaller color number first.
@zipforth  If you want to show the effect you want, you can add a new material 5 and change the background color of the object to material 5, as shown in the figure.

zhimin-zeng-bambulab, 2024-02-04:
@zipforth  Did the above method solve your problem?

zipforth, 2024-02-04:
Thanks, I edited the stl to fix the overlap and it worked!

closed 2024-02-05 by zhimin-zeng-bambulab