bambulab/BambuStudio #3306: Exhaust fan control is missing from Filament Settings in v1.8.2.56

tsmith35, 2024-01-04:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 22H2

### Additional system information

Hardware-independent

### Printer

Bambu Lab X1C

### How to reproduce

With selected printer set to "Bambu Lab X1 Carbon 0.4 nozzle" and selected filament set to "Bambu PLA Basic", open filament settings. Click the search icon. Type "exhaust" in the search term input box. Three results will be shown:

Exhaust fan : Fan speed
Exhaust fan : Fan speed  <-- yes, it is listed twice
Exhaust fan : Activate air filtration

Clicking on any of the results will cause the filament Cooling tab to be focus. That tab has options for three categories:

Cooling for specific layer
Part cooling fan
Auxiliary part cooling fan

There are no settings for the exhaust / chamber fan.

[image]

[image]

### Actual results

No possibility of controlling exhaust/chamber fan. I tested this at both HD (1920x1080) and 4k (3840x2160) resolutions on three different computers. No differences noted.

### Expected results

Exhaust/chamber fan settings should appear.

### Project file & Debug log uploads

Specific to software only.

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GuzziRaz, 2024-01-05:
You need to edit the Printer settings and enable "Support Air Filtration" in order for the exhaust fan to show up in filament settings. When I complained about that the answer was it's is only for X1E. I can't really understand why... So we have a fan but no control for it other than editing G-code.

tsmith35, 2024-01-05:
> You need to edit the Printer settings and enable "Support Air Filtration" in order for the exhaust fan to show up in filament settings. When I complained about that the answer was it's is only for [...]

Thank you for the concise answer! I guess I'm only a peon, since I only have an X1C. lol

I agree with you; why provide an exhaust fan, but not provide any control? I'm guessing it wouldn't take much for the Bambu Lab programmers to add this to Bambu Studio. OrcaSlicer imposes no such silly restrictions. 

This reminds me of the crazy struggle to get Bambu Lab's firmware guys to "allow" people who upgraded from P1P to P1S to use their printers as genuine P1S printers instead of pimped-up P1P printers. It took months of campaigning to get them to add it. 

Thank you for pointing me in the right direction.

XunZhangBambu, 2024-01-05:
@tsmith35  For X1C,we have default exhaust fan setting in filament start gcode.Why we don't default enable the "support air filtration" is because we think the default exhaust fan setting is quite fine for printing. If you are not satisfied with the default then you can enable the option and modify it. 
[image]

tsmith35, 2024-01-05:
> @tsmith35 For X1C,we have default exhaust fan setting in filament start gcode.Why we don't default enable the "support air filtration" is because we think the default exhaust fan setting is quite [...]

Wouldn't it be better to permit editing the exhaust fan speed from within Bambu Studio? Many people, including myself, do not like the fan to run at full speed, especially when printing ABS, ASA, PC or other filaments that need a warm environment to print well. What is most confusing is that all of these printers fully support air filtration: X1E, X1C, P1S -- but only X1E has the **_Activate air filtration_** option set true by default.

It also doesn't make sense to build and advertise all three models as supporting air filtration, but only enable default control of the fan speed for the X1E. While user opinion of what is acceptable may differ with respect to exhaust fan speed control, all three printers are fully capable of supporting fan speed setting. Making X1C and P1S users resort to G-code in order to edit fan speeds is somewhat pointless, since the X1C and P1S can both benefit from the existing X1E fan speed controls. The only change required is to add one line to **Bambu Lab X1.json** and **Bambu Lab P1S.json**:

_`    "support_air_filtration": "1",`_

Additionally, even the X1E suffers from incomplete implementation; only **Bambu Lab X1E 0.4 nozzle.json** has the above option set true by default. The parameter should have been set from within **Bambu Lab X1E.json** to avoid unnecessary complexity. The exhaust fan capability is specific to the machine, not the machine + nozzle size, so the option default should be located in the machine parameter set.

I hope you understand my reasoning. If fan speed control was thought to be necessary for the X1E (and only with the 0.4 nozzle), shouldn't the X1C and P1S benefit from said control? It only requires two lines of code to be added.

XunZhangBambu, 2024-01-05:
I understand your reasoning.
1. Why does only the "Bambu X1E 0.4 nozzle.json" have the "support_air_filtration" ?
Our presets have an inheritance mechanism.You can see in the "inherits"  param. According to the inheritance structure, the 0.2 nozzle, 0.6 nozzle, and 0.8 nozzle will inherit from the 0.4 nozzle. This mechanism helps reduce repetitive fields in the JSON file.

2. Why is it not enabled by default?
X1C and X1E share most of the material presets. However, the exhaust fan parameters here are designed to enhance filtration rather than adjust the chamber's cooling. Considering that modifying these parameters can control the chamber fan of other printers, we've kept this option for other models as well but without enabling it by default. For experienced users who have the request, adjustments can be made through this option.

I think the wording can be revised. For example, "Support air filtration enhancement" may be better.

tsmith35, 2024-01-05:
> 1. Why does only the "Bambu X1E 0.4 nozzle.json" have the "support_air_filtration" ?
>    Our presets have an inheritance mechanism.You can see in the "inherits"  param. According to the inheritance [...]

Thank you for the response. I understand the inheritance, but it seems incorrect to associate the printer air filtration with specific nozzle diameters. Just personal preference. I'll get over it. lol

> 2. Why is it not enabled by default?
>    X1C and X1E share most of the material presets. However, the exhaust fan parameters here are designed to enhance filtration rather than adjust the chamber's [...]

If it's just a matter of semantics, perhaps the X1C and P1S can be labeled differently. For the X1E, the label could be **Enable air filtration** and for the X1C and P1S, it could be called simply **Chamber fan control**, with the parameter defaulting to on and the default fan speed for said printers being set to whatever the current default chamber fan speed is now. Such a change would be trivial to implement and would avoid unnecessary effort for your customers. Bambu Lab printers are marketed as easy to use; why hide this feature behind a parameter menu that many users will never see? Ease of use is paramount.

I can't tell you how many times I've seen people asking how to slow down their exhaust fans. I myself simply went into the machine G-code to slow it down, but that doesn't mesh well with Bambu Lab's efforts to provide a great user experience. While I later discovered the option was hidden away in the submenu for machine configuration, there's really no reason for it to be that way. If someone is willing to pay $1500+ to get a Bambu Lab X1C, why should basic chamber fan speed control be hidden?

Again, it's a trivial change that would make the Bambu Lab customer experience better. Is there a downside to making this small change?

XunZhangBambu, 2024-01-05:
You provide an idea to sperate the chamber fan control from "exhaust fan control for air filtration" in X1E .

Downside:
This should be an option, we do not want inexperienced users to casually adjust the fan speed, as it could potentially lead to print quality issues or even print failures. Therefore, the best approach would be to place it within the filament_start_gcode, overriding the default speed when activated. However, implement it this way may cause failures in slicing 3MF files generated by the new version of bambu studio with the old version.

Anyway, I am also considering an alternative approach.

GuzziRaz, 2024-01-05:
> I think the wording can be revised. For example, "Support air filtration enhancement" may be better.

1. For P1/X1C it could be called "Support chamber evacuation" as that is what many people call it. Lota of people struggle with Machine End G-code to run the chamber fan for some time after a print has finished.
2. Wording aside, the functionality should be also revised: All printers with a chamber fan and a chamber temperature sensor should obviously have, and honor, a target chamber temperature. Too cool = fan off. Too hot = full speed. Anything in between = PID regulation. This reduce noise. Noise is bad, heuristics is good.

capsel22, 2024-01-05:
Thanks for logging this issue. I have learned something new.
Should I enable this on my P1S? Sorry im a newbie.
[image]

tsmith35, 2024-01-05:
> You provide an idea to sperate the chamber fan control from "exhaust fan control for air filtration" in X1E .
> 
> Downside: This should be an option, we do not want inexperienced users to casually adjust the fan speed, as it could potentially lead to print quality issues or even print failures.

I'm an engineer, and I've worked several places where we've purchased 3D printers for use by the engineering staff. My observation is that everybody gets excited when the printer arrives, but few people use it regularly. Unless the users print at home, the skill level is fairly low. Those are the expected X1E users.

Contrast that with the X1C and P1S. $1000-1500 is a fairly large expense for a hobby, so the user base is more likely to be at least semi-skilled at operating their printers. Many are quite skilled, especially those who make money by printing. Ironically, the X1C and P1S users are denied the convenience of having full fan control by default. 

If lack of skill is truly a concern, perhaps chamber fan control could be turned on by default (and at default levels) for the X1C and P1S, but users would not be shown the speed controls unless they select Advanced mode or Developer mode. 

Realistically, that should probably be the case with the X1E as well; you're making an assumption that average X1E users are more advanced than average X1C/P1S users. The converse is much more likely to be true.

tsmith35, 2024-01-22:
@XunZhangBambu okay, so perhaps a different approach would be more successful...

Is there any functionality within Bambu Studio to facilitate user overrides? For example, if I would like to override or add support_air_filtration to the system machine profile "Bambu X1C 0.4 nozzle.json", I presently have to edit that file in both "C:\Program Files\Bambu Studio\resources\profiles\BBL\machine" and the one in "AppData\Roaming\BambuStudio\system\BBL\machine" to add the support_air_filtration:1 line.

That works fine, but I have to do that every time I update Studio. I have semi-automated the process using a grep-type utility, but it would be preferable to have an overrides file or files to help avoid this kludgy solution.

GuzziRaz, 2024-01-22:
> Is there any functionality within Bambu Studio to facilitate user overrides?

I had some wild ideas in #2831. I believe the current “inherits xx” has everything needed, we’d just need a set of user’s shim files that we are allowed to edit - and a little bit of support for them in BS.

tsmith35, 2024-01-22:
> > Is there any functionality within Bambu Studio to facilitate user overrides?
> 
> I had some wild ideas in #2831. I believe the current “inherits xx” has everything needed, we’d just need a set of user’s shim files that we are allowed to edit - and a little bit of support for them in BS.

I like it! That way, the defaults can be dumbed down so new users don't get confused and mess things up, but experienced users can do what is needed to get the results they want.

Nicog77, 2024-02-23:
This is fix in the bambu studio v1.8.4.51. You have to modify you printer setting and tick the Enable Active air filtration option.
Then you can add this option in the filament window.