bambulab/BambuStudio #3307: Crash on activating Variable Layer Height

caco3, 2024-01-05:
### Bambu Studio Version

BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 23.10

### Printer

P1S

### How to reproduce

1. Load model (eg. [M8.zip](https://github.com/bambulab/BambuStudio/files/13841407/M8.zip))
2. Activate Variable Layer Height

### Actual results

Speicherzugriffsfehler (Speicherabzug geschrieben)
(Memory access error)

### Expected results

no crash

### Project file & Debug log uploads

```
2024-01-05 12:29:23.351660[Thread 0x00007fb233c32c80]:gui mode, Current BambuStudio Version 01.08.02.56
2024-01-05 12:29:23.377491[Thread 0x00007fb233c32c80]:load_language: language , initial: 1
2024-01-05 12:29:23.377534[Thread 0x00007fb233c32c80]:language provided by BambuStudio.conf: en_GB
2024-01-05 12:29:23.377653[Thread 0x00007fb233c32c80]:Switching wxLocales to en_GB
2024-01-05 12:29:23.398766[Thread 0x00007fb233c32c80]:Update_dark_mode_flag: switch the current dark mode status to 0 
2024-01-05 12:29:23.402753[Thread 0x00007fb233c32c80]:begin to show the splash screen...
2024-01-05 12:29:23.504219[Thread 0x00007fb233c32c80]:loading systen presets...
2024-01-05 12:29:23.509200[Thread 0x00007fb233c32c80]:[BBL Updater]:Checking whether the profile from resource is newer
2024-01-05 12:29:23.515249[Thread 0x00007fb233c32c80]:initialize_network_module, successfully loaded library /home/***/.config/BambuStudio/plugins/libbambu_networking.so, module 0x55c11f22f6f0
2024-01-05 12:29:23.515337[Thread 0x00007fb233c32c80]:get_network_function, can not find function bambu_network_get_profile_3mf
2024-01-05 12:29:23.515360[Thread 0x00007fb233c32c80]:on_init_network: on_init_network, load dll ok
2024-01-05 12:29:23.515365[Thread 0x00007fb233c32c80]:get_network_agent_version=01.08.02.05
2024-01-05 12:29:23.515368[Thread 0x00007fb233c32c80]:on_init_network: on_init_network, compatibility version
2024-01-05 12:29:23.524713[Thread 0x00007fb233c32c80]:on_init_network, create network agent...
2024-01-05 12:29:23.525216[Thread 0x00007fb233c32c80]:NetworkAgent, this 0x55c11f212cf0, network_agent=0x55c11f212cd0, create_agent_ptr=1, log_dir=/home/***/.config/BambuStudio
2024-01-05 12:29:23.529489[Thread 0x00007fb233c32c80]:init_networking_callbacks: enter, m_agent=0x55c11f212cf0
2024-01-05 12:29:23.531797[Thread 0x00007fb233c32c80]:init_networking_callbacks: exit, m_agent=0x55c11f212cf0
2024-01-05 12:29:23.532291[Thread 0x00007fb233c32c80]:load_presets enter, substitution_rule 3, preferred printer_model_id 
2024-01-05 12:29:23.532329[Thread 0x00007fb233c32c80]:load_vendor_configs_from_json enter, path /home/***/.config/BambuStudio/system, compatibility_rule 2
2024-01-05 12:29:23.533337[Thread 0x00007fb233c32c80]:load_vendor_configs_from_json: parse /home/***/.config/BambuStudio/system/BBL.json, got description:  "the initial version of BBL configurations"
2024-01-05 12:29:23.653245[Thread 0x00007fb233c32c80]:load_user_presets521 entry and user is: 246618111
2024-01-05 12:29:23.653282[Thread 0x00007fb233c32c80]:load_presets enter, load presets from "/home/***/.config/BambuStudio/user/246618111/process", current type process
2024-01-05 12:29:23.653460[Thread 0x00007fb233c32c80]:load_presets enter, load presets from "/home/***/.config/BambuStudio/user/246618111/filament/base", current type filament
2024-01-05 12:29:23.654056[Thread 0x00007fb233c32c80]:load_presets1220 load config successful and preset name is:Purefil PETG Basic @Bambu Lab P1S 0.4 nozzle
2024-01-05 12:29:23.654306[Thread 0x00007fb233c32c80]:load_presets enter, load presets from "/home/***/.config/BambuStudio/user/246618111/filament", current type filament
2024-01-05 12:29:23.654551[Thread 0x00007fb233c32c80]:load_presets1220 load config successful and preset name is:Purefil PETG Basic Schwarz
2024-01-05 12:29:23.654764[Thread 0x00007fb233c32c80]:load_presets enter, load presets from "/home/***/.config/BambuStudio/user/246618111/machine", current type machine
2024-01-05 12:29:23.654818[Thread 0x00007fb233c32c80]:update_compatibility for all presets enter
2024-01-05 12:29:23.654864[Thread 0x00007fb233c32c80]:update_compatibility for all presets exit
2024-01-05 12:29:23.654869[Thread 0x00007fb233c32c80]:update_compatibility for all presets enter
2024-01-05 12:29:23.654887[Thread 0x00007fb233c32c80]:update_compatibility for all presets exit
2024-01-05 12:29:23.659790[Thread 0x00007fb233c32c80]:update_compatibility for all presets enter
2024-01-05 12:29:23.660393[Thread 0x00007fb233c32c80]:update_compatibility for all presets exit
2024-01-05 12:29:23.660414[Thread 0x00007fb233c32c80]:load_presets finished, returned substitutions 0
2024-01-05 12:29:23.660426[Thread 0x00007fb233c32c80]:create the main window
2024-01-05 12:29:23.808243[Thread 0x00007fb233c32c80]:CreateWebView: file:/tmp/.mount_BambuSbTFEf4/resources/web/homepage/index.html?lang=en_US
2024-01-05 12:29:24.159176[Thread 0x00007fb233c32c80]:select_technology: change the printer technology from 2 to 0
[... 423 lines cut ...]
2024-01-05 12:30:28.896515[Thread 0x00007fb233c32c80]:_save_model_to_file:5801, before add relation file to 3mf
2024-01-05 12:30:28.896561[Thread 0x00007fb233c32c80]:_save_model_to_file:5823, finished exporting 3mf
2024-01-05 12:30:29.126640[Thread 0x00007fb233c32c80]:update_print_volume_state, print_volume {0, 0, 0} to {256, 256, 250}, got 1 printable istances
2024-01-05 12:30:29.126710[Thread 0x00007fb233c32c80]:update_background_process: enter, force_validation=0 postpone_error_messages=0, switch_print=1, was_running=0
2024-01-05 12:30:29.126742[Thread 0x00007fb233c32c80]:update_print_volume_state, print_volume {0, 0, 0} to {256, 256, 250}, got 1 printable istances
2024-01-05 12:30:29.126760[Thread 0x00007fb233c32c80]:get_current_slice_result:m_current_plate 0, current_plate 0x55c1268256a0
2024-01-05 12:30:29.129311[Thread 0x00007fb233c32c80]:apply, Line 1019: enter
2024-01-05 12:30:29.129385[Thread 0x00007fb233c32c80]:apply, got changed_keys, size=1
2024-01-05 12:30:29.129409[Thread 0x00007fb233c32c80]:apply, i=0, key=enable_prime_tower
2024-01-05 12:30:29.130321[Thread 0x00007fb233c32c80]:apply 1575: finished,  this 0x55c11efc5ab0, m_modified_count 1, apply_status 0, m_support_used 0
2024-01-05 12:30:29.130686[Thread 0x00007fb233c32c80]:update_background_process: background process apply result=0
2024-01-05 12:30:29.130709[Thread 0x00007fb233c32c80]:update_background_process, Line 4893: was_running = 0, running 0, invalidated=0, return_state=0, internal_cancel=0
2024-01-05 12:30:29.130727[Thread 0x00007fb233c32c80]:update_background_process, Line 4923: background data valid, return_state=0
2024-01-05 12:30:29.130739[Thread 0x00007fb233c32c80]:get_enable_print_status: m_print_select 1, enable= 0 
2024-01-05 12:30:29.130750[Thread 0x00007fb233c32c80]:get_enable_slice_status: m_slice_select 1, enable= 1 
2024-01-05 12:30:29.130758[Thread 0x00007fb233c32c80]:update_slice_print_status m_slice_select 1: can_slice= 1, can_print 1, enable_slice 1, enable_print 0 
2024-01-05 12:30:29.130772[Thread 0x00007fb233c32c80]:update_background_process, Line 4966: exit, return_state=0
2024-01-05 12:30:29.130782[Thread 0x00007fb233c32c80]:restart_background_process, Line 5004: not started
2024-01-05 12:30:30.602221[Thread 0x00007fb233c32c80]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
```

### Checklist of files to include

- [X] Log file
- [X] Project file

aleung, 2024-01-06:
Same issue on Ubuntu 22.04 with AppImage.

BambuStudio crashes immediately when I click the "Variable Layer Height" button, without the dialog popping up. It always crashes, regardless of the model.

Haidiye01, 2024-01-08:
have you try bambuStudio in windows 10 or windows 11?Is it ok?

caco3, 2024-01-08:
> have you try bambuStudio in windows 10 or windows 11?Is it ok?

Yes, it works well on Windows 10.

Haidiye01, 2024-01-10:
[image]
i test ok in Unbantu  
[image]

aleung, 2024-01-14:
Just found that the same issue had been fixed in PrusaSlicer 2.6.0 : https://github.com/prusa3d/PrusaSlicer/issues/7274

The issue is OpenGL related. PrusaSlicer fixed it by using OpenGL 3.3.

I confirm that BambuStudio doesn't crash when disable hardware-accelerated by starting with environment variable `LIBGL_ALWAYS_SOFTWARE=1`:
`LIBGL_ALWAYS_SOFTWARE=1 ./PrusaSlicer-2.4.0-beta1+linux-x64-202110251331.AppImage`

aleung, 2024-01-14:
>  i test ok in Ubuntu 

@Haidiye01 
According to https://github.com/prusa3d/PrusaSlicer/issues/8756#issuecomment-1228326594, the issue is with some Intel Mesa drivers. So it might only happen on users who use Intel integrated graphics.

caco3, 2024-01-14:
I can confirm that `LIBGL_ALWAYS_SOFTWARE=1` fixes it.

And I also haven an Intel GPU using the Mesa driver:
```bash
~$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
~$ glxinfo | grep -i vendor
server glx vendor string: SGI
client glx vendor string: Mesa Project and SGI
    Vendor: Intel (0x8086)
OpenGL vendor string: Intel
```

caco3, 2024-01-25:
This seems to be fixed in `v01.08.04.51_20240117`, so I am going to close this issue.

aleung, 2024-01-25:
@caco3 Version `1.8.4` still crashes on my Ubuntu. Did you leave `LIBGL_ALWAYS_SOFTWARE=1` in your environment?

caco3, 2024-01-25:
@aleung I have 2 systems, desktop and laptop, before 1.8.4 it was crashing on both systems (without `LIBGL_ALWAYS_SOFTWARE=1`), but now it only crashes on the laptop. I reopen the issue because I only tested on my desktop system (explicitly without `LIBGL_ALWAYS_SOFTWARE=1`). So maybe the non-crash is caused by something else.

caco3, 2024-08-29:
I am closing this issue since the issue seems to be solved in the never versions.

closed 2024-08-29 by caco3

Haidiye01, 2024-08-29:
ok,i have received your letter ,thank you!

ladyluck9702, 2025-05-17:
I am having this issue with my new lap top that is running windows 11. I am not understanding what is is that they did to fix their issue. Every time I click the Variable Layer Height button is shuts down the Bambu Studios. Can you explain what it is I need to do to fix this? I did not have this issue on my old lap top running windows 10.