bambulab/BambuStudio #3309: Home page/menus not loading.

TheIndiglo, 2024-01-05:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Additional system information

         Operating System: Windows 11 Home 64-bit (10.0, Build 22631) (22621.ni_release.220506-1250)
                 Language: English (Regional Setting: English)
      System Manufacturer: ASUSTeK COMPUTER INC.
             System Model: ASUS TUF Gaming F17 FX706HCB_FX706HCB
                     BIOS: FX706HCB.313 (type: UEFI)
                Processor: 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz (16 CPUs), ~2.3GHz
                   Memory: 16384MB RAM
      Available OS Memory: 16122MB RAM
                Page File: 16946MB used, 5063MB available
              Windows Dir: C:\WINDOWS
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
         User DPI Setting: 96 DPI (100 percent)
       System DPI Setting: 96 DPI (100 percent)
          DWM DPI Scaling: Disabled
                 Miracast: Available, with HDCP
Microsoft Graphics Hybrid: Supported
 DirectX Database Version: 1.4.7
           DxDiag Version: 10.00.22621.0001 64bit Unicode

### Printer

X1 Carbon

### How to reproduce

Click on the "Home" button.
Open any menus. 

### Actual results

Never loads, just remains a blank page. 

### Expected results

Open model "store". 

### Project file & Debug log uploads

No specific file, just the actual software. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-01-08:
To clearly explain the problem, please provide application logs (ref the wiki). It is also best to provide software screenshots or record a video of the problem.

https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

AlexNoop, 2024-01-30:
hello,

same problem here ;)
blank home and verry long startup

[debug_Tue_Jan_30_07_16_30_6756.log.0.txt](https://github.com/bambulab/BambuStudio/files/14093682/debug_Tue_Jan_30_07_16_30_6756.log.0.txt)

[image]
[image]
[image]

bambu123, 2024-01-30:
@AlexNoop 
I'm very sorry for the inconvenience.
Long startup time may be the root cause, causing WebView to timeout.
In order to analyze the problem of long startup time, it is necessary to manually capture the program DUMP. Please refer to the following wiki for help in providing DMP files.

AlexNoop, 2024-02-04:
hey ! thanks for reply. dmp files abrove 1gb ....

bambu123, 2024-02-04:
> hey ! thanks for reply. dmp files are there : https://gofile.me/74DmP/lI3Fj0uTs

Sorry, failed to download. I'll retry later.

bambu123, 2024-02-06:
> hey ! thanks for reply. dmp files are there : https://gofile.me/74DmP/lI3Fj0uTs

Sorry, can't download, please check.

fahaaaxiki, 2024-09-27:
https://github.com/bambulab/BambuStudio/issues/3807 Could this issue help you?