bambulab/BambuStudio #3313: Titlebar visible on Linux X11

hadess, 2024-01-05:
### Bambu Studio Version

git

### Where is the application from?

Built from source by yourself or others

### OS version

Flathub GNOME SDK 45

### Printer

N/A

### How to reproduce

Launch BambuStudio

### Actual results

It has 2 titlebars, the window manager one, and the application provided one.

[image]

### Expected results

Just one titlebar.

I've managed to hack this changing the value of `BORDERLESS_FRAME_STYLE` in `src/slic3r/GUI/MainFrame.cpp` to `(wxRESIZE_BORDER | wxBORDER_SIMPLE)` but that makes the minimise and maximise buttons not work (the minimise button shouldn't appear on my GNOME system), and the window isn't resizable

[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-01-08:
Windows has done special processing. But we don't have much energy to optimize Linux version.

hadess, 2024-06-26:
> Orca Slicer inherits that too, unfortunately.

https://github.com/SoftFever/OrcaSlicer/issues/4411

celevra, 2024-07-19:
got that on appimage and flatpack

lucasfalcao3d, 2026-05-27:
> If anyone comes across this: The double title bar really bothered me, so I added an option (Preferences / General Settings) to disable the custom title bar in favor of the OS native title bar and move all menus to a new menu button.
> <img alt="Image" width="2000" height="834" src="https://private-user-images.githubusercontent.com/32908927/582978182-70e82208-dd59-426d-9204-1081876fd8e5.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Nzk4NDU3MTUsIm5iZiI6MTc3OTg0NTQxNSwicGF0aCI6Ii8zMjkwODkyNy81ODI5NzgxODItNzBlODIyMDgtZGQ1OS00MjZkLTkyMDQtMTA4MTg3NmZkOGU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA1MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwNTI3VDAxMzAxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMwZGFjZWE3M2RhMjBmZDg5NWQ3YjM3Mjg5ZTA2NTU4Y2VmNDU4NjU5OTE0ZjFiOTE2ODNjMWRhMDQxODMzOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRnBuZyJ9.S9QK4awblOpBc12uLidHcyM4C0bW3zOl4C7Qugk_5-g">
> 
> I will try to create a pull request to get this upstream, but I want to validate first that I didn't break the Windows (and maybe Mac) version, but getting this compiled on Windows is super annoying. In the meantime you can get a updated flatpak from [here](https://github.com/jonpetersathan/BambuStudio/releases/tag/v02.06.00.51A). Or if you don't trust my flatpak, you can build [it](https://github.com/jonpetersathan/BambuStudio) yourself with the instructions [here](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide).

This is a very good proposal, IMHO it would be better if it uses the green color accent as a selected element. It also needs to incorporate the Calibration menu, when _Develop mode_ is enabled or the Calibration menu should be in another drop down menu? 

I've made a kind just like this [here](https://github.com/bambulab/BambuStudio/issues/10016).

lucasfalcao3d, 2026-08-19:
Hello @hadess, is there any chance to add this integration, that @jonpetersathan created on the Linux releases at some point?

hadess, 2026-08-19:
> Hello [@hadess](https://github.com/hadess), is there any chance to add this integration, that [@jonpetersathan](https://github.com/jonpetersathan) created on the Linux releases at some point?

Not in the state it seems to be in (hard to know, as there's no upstream merge request for the changes that I know of).

Before it could be merged, it needs to be always enabled (not an option), working on both X11 and Wayland (although I understand that Wayland support in other parts of the application isn't complete), and it needs to function as a titlebar, which means showing the title (the name of the currently open model) and allowing it to be dragged by its titlebar.

lucasfalcao3d, 2026-08-20:
I've made two mock up suggestions, if someone are willing to work on it.

- Option 1: A more clean menu design like suggested by @jonpetersathan, I've added a divider to separate the menu, because it's a dropdown menu and not a tab. All the menu go inside the hamburger icon and looks super clean.

- Option 2: A more traditional menu and more user friendly, because most apps use this way.

[image]

[image]

lucasfalcao3d, 2026-09-16:
Hey @hadess , if possible, there is any chances to review the implementation Sproctor did before the next release? 🥹