bambulab/BambuStudio #3315: Load of GTK errors on startup

hadess, 2024-01-05:
### Bambu Studio Version

git

### Where is the application from?

Built from source by yourself or others

### OS version

Flathub GNOME SDK 45

### Printer

N/A

### How to reproduce

Launch BambuStudio from Flatpak (https://github.com/flathub/flathub/pull/4822)

This build uses wxWidgets 3.2.4 rather than 3.1.5 in the appimage because:
- I could not get BambuStudio to build with 3.1.5
- the AppImage build throws even more messages, and those messages are worse (widgets cast from one class to an incompatible class for example).

### Actual results

[bambustudio-gtk-errors-flatpak.txt](https://github.com/bambulab/BambuStudio/files/13847155/bambustudio-gtk-errors-flatpak.txt)

### Expected results

No errors.

### Project file & Debug log uploads

For completeness sake, the AppImage errors, on Fedora 39 and `gtk3-3.24.39-1.fc39.x86_64`

[bambustudio-gtk-errors-appimage.txt](https://github.com/bambulab/BambuStudio/files/13847186/bambustudio-gtk-errors-appimage.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hadess, 2024-01-09:
> what is the behavior on the screen?
> 
> is the splash screen or main window popup?

It works...

caco3, 2024-01-09:
same warnings on the official release image on Ubuntu

hadess, 2024-08-06:
Poke

seanlanders92, 2024-08-21:
Just downloaded on a fresh install of ubuntu 22.04 using latest flatpak build when attempting to run ``` flatpak run com.bambulab.BambuStudio ``` process times out and due to module not found canberra-gtk-module I have installed canberra gtk and issue wtill persists

Lrapava, 2025-12-06:
Experiencing the same issue on Fedora with flathub..