bambulab/BambuStudio #3316: Bambu Studio generated 3MF files cannot be opened with other slicers

mcmaven, 2024-01-06:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1. Use PrusaSlicer to open a Bambu Studio generated 3MF file.
2. Use Cura to open a Bambu Studio generated 3MF file.

### Actual results

1. PrusaSlicer generates an error message "Prusa Slicer has encountered an error. Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 18".
2. Cura generates an error message "No models in file".

### Expected results

I expected the Bambu Studio generated 3MF file to be opened by the other slicers.
This is an extremely simple file with only one part and no options set.
I used the same part file (18650 Battery 4-3 D-Cell Spacer.STL) to generate 3MF files using Cura and PrusaSlicer and Bambu Studio was able to successfully open both files.
If the Project ("Print Profile") files generated by Bambu Slicer are not compliant with the 3MF file format specification (https://github.com/3MFConsortium/spec_core/releases/download/1.3.0/3MF_Core_Specification_v1.3.0.pdf), then it should not use the 3MF extension!

### Project file & Debug log uploads

The Bambu Studio .3MF file - 
[18650 Battery 4-3 D-Cell Spacer Test.zip](https://github.com/bambulab/BambuStudio/files/13847740/18650.Battery.4-3.D-Cell.Spacer.Test.zip)

There is no Bambu Studio .log file for this project.

The STL file - 
[18650 Battery 4-3 D-Cell Spacer.STL.zip](https://github.com/bambulab/BambuStudio/files/13847743/18650.Battery.4-3.D-Cell.Spacer.STL.zip)

PrusaSlicer and Cura 3MF files - 
[18650 Battery 4-3 D-Cell Spacer Test Cura and PrusaSlicer 3MF files.zip](https://github.com/bambulab/BambuStudio/files/13847746/18650.Battery.4-3.D-Cell.Spacer.Test.Cura.and.PrusaSlicer.3MF.files.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DanBao-Bambu, 2024-01-08:
The current default save and save as operations in Bambu Studio produce 3MF files based on the 3MF Consortium's [3MF Production Extension](https://github.com/3MFConsortium/spec_production/blob/1.1.2/3MF%20Production%20Extension.md) specification.

If you wish to export 3MF files compliant with the 3MF Core Specification for opening in Cura and PrusaSlicer, you can achieve this by using the "Export Generic 3MF" operation.
[image]

mcmaven, 2024-01-18:
Hi,

Thanks for the information.

FYI, I did a test where I opened the "Triple-axis Tourbillon Model for Kit
007.3MF" on page https://makerworld.com/en/models/13297#profileId-99236 in
Bambu Studio 1.8.4.51, then exported the project as a 'generic' and a Bambu
3MF file. I think that this is a very complicated 3MF.

PrusaSlicer 2.7.1 could not open either exported file.
Cura 5.6.0 opened the generic file but could only generate a partial slice
of Plate 1. And it could not open the exported Bambu 3MF file.
3D Viewer could open both the generic and Bambu 3MF files.
3D Builder opened the generic file but could not open the exported Bambu
3MF file.

Regards,

Steve

DanBao-Bambu, 2024-01-22:
I have reviewed the 3mf file in the link you provided and also exported a universal 3mf file. I found that the loading error line is as follows. 
[image]
Found the original model definition based on the object ID index and found that the model is "other". Perhaps this is the reason why it cannot be opened.
[image]

mcmaven, 2024-01-22:
Hi DanBao,

Thank you for taking the time to analyze this. What program do you use to
display the content of the 3MF file?

Regards,

Steve

DanBao-Bambu, 2024-01-22:
3mf files can be decompressed, and you can use decompression tools to decompress them. The model data of 3mf is saved in the "/3D" folder, and "3dmodel. model" can be opened using text tools. It is used to save all model files in the General 3mf file and to save the index of sub files in the Production Extension specification (these sub files are in the/3D/Objects folder).
We will try to solve the problem you mentioned earlier, but it will take some time for research. Thank you for your question.

DanBao-Bambu, 2024-01-22:
When you export a generic 3mf file, all model information is in the "/3D/3dmodel. model" file, and you can open it like opening a ". txt" file. Then search for the keywords you need. If you want to view object information, search for "<object id=" ".

dsanders2, 2024-07-19:
Other slicers can open other slicers .3mf files with no special actions needed. It would be best to use a different file format so not to confuse a Bambu Studio .3mf and all other "generic" .3mfs. I get that it still follows standards for .3mf formatting. However, if it is exclusive to just Bambu Studio it should have it's own file extension. This type of exclusivities only hurts the community. If this is not a bug it was a lousy design choice.

SaltWei, 2024-07-19:
[image]
We have submitted the code to Prusa and Cura to support the 3MF extension protocol. Prusa has already accepted our submission, but Cura has not done so yet.

SaltWei, 2024-07-19:
> > This type of exclusivities only hurts the community.
> 
> ![image](https://private-user-images.githubusercontent.com/110660407/350260589-c19c47b5-7e77-41e9-81b9-af5462247c9e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzNjk1ODQsIm5iZiI6MTcyMTM2OTI4NCwicGF0aCI6Ii8xMTA2NjA0MDcvMzUwMjYwNTg5LWMxOWM0N2I1LTdlNzctNDFlOS04MWI5LWFmNTQ2MjI0N2M5ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxOVQwNjA4MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04OWYyZjNhYTEzNGY3Yzk5OWJlYmQ2YWI2ZTllNWY3ZGViOWQ2MGE2YzY3NzZiYTE0ZmY5NDk0ZGFkYTQ3NDlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.wlQDrdaNFi8-xFXNe3w2iOV_rN6y37Vs-F_563GuIS0) We have submitted the code to Prusa and Cura to support the 3MF extension protocol. Prusa has already accepted our submission.

Bambu does not have exclusivity.

mcmaven, 2024-07-19:
I save Bambu 3MF files with a '.bambu.3mf' extension to identify them.

raistlinJ, 2025-08-09:
I created a converter here: https://github.com/raistlinJ/3mf_bambu2prusa
Tested with several models and it seems to work; even transfers painted objects.

mcmaven, 2025-08-12:
Does the code strip out the extra Bambu info?

mcmaven, 2025-08-13:
> yes. I built the tool by reversing the format + trial/error, so I only
> kept info that was required for prusa to load the model.

How is it different using the Studio "File/Export/ Export Generic 3MF..."
option?

mcmaven, 2025-08-13:
> A generic 3mf doesn't have paint, as far as I know/observed. Additionally,
> in some cases I still wasn't able to open an exported generic 3mf in prusa
> (I think one of the issues is the object types; prusa doesn't seem to like
> non-"mesh" types; bambu has "other" types). Again, just did most of this
> through observation so take this info with a grain of salt.

A while back I complained to Bambu that they shouldn't use a .3MF extension
if their 3MF files couldn't be imported by Prusa Slicer, Cura, etc. Then I
dug into the 3MF file format and realized that Bambu 3MF files were
'compliant' with the 3MF specification, which allows all sorts of public
and proprietary extensions. The problem was that Bambu 3MF files included
information for features/extensions that the other slicers didn't support,
e.g., Multiple plates. So in theory it was a slicer problem, not a 3MF
problem. A 'generic' 3MF is really a subset of a Bambu 3MF, i.e., the
lowest common denominator. I wasn't able to find formal documentation on
Plate, Paint, or Support extensions as described by the 3MF Consortium's 3MF
Production Extension
<https://github.com/3MFConsortium/spec_production/blob/1.1.2/3MF%20Production%20Extension.md>
specification.
Do you know if they exist?

Looking at a Bambu 3MF file, since it's really just a bunch of XML files
after you unZIP it, you could figure out how they defined their extensions,
but that was more than I was willing to take on.