bambulab/BambuStudio #3318: Bambu Studio crashes when trying to slice project

GabrielQuintana, 2024-01-06:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home

### Additional system information

CPU: i9-14900k
Memory: 64 GB 6400mhz
GPU: Asus Nvidia GeForce RTX 4070 Ti

### Printer

Bambu Lab X1C

### How to reproduce

Open the provided 3mf file in bambu studios, and click "slice plate"

### Actual results

Bambu Studio crashes

### Expected results

The plate should be successful sliced

### Project file & Debug log uploads

[Log and Project.zip](https://github.com/bambulab/BambuStudio/files/13848627/Log.and.Project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-01-08:
We found that some models cannot be sliced on Win 11. We'll try to fix this in next big version.

QingZhangBambu, 2024-01-25:
[image]

QingZhangBambu, 2024-01-25:
@GabrielQuintana  is that still happend in the latest version on your PC?

QingZhangBambu, 2024-01-26:
@GabrielQuintana holle, there is a method that some people said is helpful, may you try it and check if it is useful?.
For Win 11, here is how to set the affinity : 
1- Open Task Manager (CTRL+SHIFT+ESCAPE)
2- Go to Details
3- find bambu-studio.exe and right click to chose "Affinity"
[image]
4- From the popup, it is possible to choose which thread(s) to use. For example, on my i9-13900K, I've got CPU0 to CPU31 (I've 8 performance cores with 2 threads, and 16 efficiency cores -> 32 "CPUs" listed)
[image]
And interestingly enough, when I slice without setting affinity (so, Bambu Studio uses all threads available) -> it crashes.
When I first set affinity to, for example, CPU0 to CPU3, or CPU0 to CPU7 -> slicing goes through successfully.

GabrielQuintana, 2024-01-26:
@QingZhangBambu Latest version (without modifying affinity) is still having the same issue where bambu studio crashes.

Though after setting the affinity to CPU0 to CPU7 (also tried CPU25-31) I got the exact same result, slicing does goes through successfully.

QingZhangBambu, 2024-01-29:
@GabrielQuintana Then when the Studio crashed?

GabrielQuintana, 2024-01-29:
@QingZhangBambu Yes, it crashes when slicing on the new version too. Should I provide logs again for the new version?

QingZhangBambu, 2024-01-29:
yes, that would be helpful

Blade-Strike, 2024-01-31:
I have the same exact issue. If I set infinity to 1-7 I'm good. Otherwise, it's a hard crash.
[sys.txt](https://github.com/bambulab/BambuStudio/files/14106181/sys.txt)

QingZhangBambu, 2024-01-31:
@Blade-Strike  maybe you can do some CPU tests. some users change the CPU and all problem be solved. https://community.intel.com/t5/Processors/i9-13900K-very-frequent-crashes-Windows-11-with-apps-games-and/m-p/1528789/highlight/true
https://github.com/bambulab/BambuStudio/issues/3354

Arcaine10, 2024-03-12:
Good morning,
I may have found a solution without having to change the affinity each time. You have to download Intel extreme tuning utility, go to advanced tuning and set performance active-core tuning to 54. Since then I have had no more problems.
[image]

GabrielQuintana, 2024-03-19:
@QingZhangBambu Sorry forgot to update here, the issue was 100% the CPU being unstable because of motherboards and intel overclocking the CPU too much. which made CPU intensive procedures like slicing crash every so often. So I'm good to close this issue if you are.

For other people experiencing the same issue as I was, here's the thread that helped me https://www.reddit.com/r/overclocking/comments/1axepvu/optimizing_stability_for_intel_13900k_and_14900k/

QingZhangBambu, 2024-03-19:
ok ;)

elyograg, 2024-03-28:
> Good morning, I may have found a solution without having to change the affinity each time. You have to download Intel extreme tuning utility, go to advanced tuning and set performance active-core tuning to 54. Since then I have had no more problems.

This worked for me.  Slicing is now stable.  I was having a ton of problems with it.  Mostly it would just crash Bambu Studio, but sometimes it would bluescreen or freeze the system so completely that the mouse pointer wouldn't even move.

masystems, 2024-05-18:
Got the same issue here, had to set it use CPU0-7. Bambu, please fix your software.

elyograg, 2024-05-20:
I did have this working with the changes to the Intel Extreme Tuning program, but it has started crashing again when slicing certain models.  I didn't change anything.

Bambu Studio also crashes slicing the same model, but it causes a bluescreen instead of crashing the program.

Skydev0h, 2024-06-03:
For me I had issue on an Intel i9-14900K, and it seems that the issue is in multithreading on both P-Cores and E-Cores.
Using a free `CoreDirector` utility I configured constraining `bambu-studio.exe` to P-Cores, and that not only fixed the crashes, but also sped up slicing by a significant margin.

[image]

evlas1013, 2025-02-11:
I'm having a similar issue but any time I go to the Preview tab, it crashes. No combination of affinity fixed it for me. i7-7700 Win10. I don't even get to the slicing step. As soon as I import a model, it crashes. If I just click the "Preview" tab, it crashes. v1.10.1.50

QingZhangBambu, 2025-02-11:
@elyograg could you post a new issue? I will ask the concerned staff to check this. your problem is not the same as this one.

MaJin1996, 2025-07-27:
你好，最新版仍然有这个问题，通过增加 CPU 亲和性解决

MaJin1996, 2025-07-27:
看了下这问题都一年半了，还没解决，高配电脑等于是不能用你们的软件，你们能不能招点正经C++ 工程师或者干脆别用C++写了。