bambulab/BambuStudio #3326: Add "filter out tiny gaps" in the strength/infil section like orcaslicer.

djmarkino94, 2024-01-07:
Is it possible to add the "filter out tiny gaps" function in the strength/infil section like in orcaslicer? very useful for removing that annoying last step to plug the holes.

djngnear, 2024-01-07:
Can you post pictures of examples please?

djmarkino94, 2024-01-08:
> Can you post pictures of examples please?

[image]

djmarkino94, 2024-01-14:
???? Its ok?

djngnear, 2024-01-14:
It already has it. Just checked. But i meant examples of how it helps in the slicing view

djmarkino94, 2024-01-14:
> It already has it. Just checked. But i meant examples of how it helps in the slicing view

Could you tell me what it's called and where it's located?

djngnear, 2024-01-14:
Yeah, it's under the Strength/Sparse Infill section. I'm using the latest version 1.8.3.89. You probably don't have Develop Mode on. Open up the preferences and it's at the bottom.

djmarkino94, 2024-01-14:
> Yeah, it's under the Strength/Sparse Infill section. I'm using the latest version 1.8.3.89. You probably don't have Develop Mode on. Open up the preferences and it's at the bottom.

Thank's so much mate

djmarkino94, 2024-01-14:
I have try, but on bambustudio don't work🥶.i mean this: this is bambustudio https://github.com/bambulab/BambuStudio/assets/111192738/991fcc63-29d5-446d-b720-a2b71de7ce76

This is orcaslicer with tiny gala activated at value 100.

https://github.com/bambulab/BambuStudio/assets/111192738/7cf2150d-5a23-4bb3-8208-74c9ebd0db91

djngnear, 2024-01-15:
I see what you mean. It doesn't work the same between the 2 slicers.

djmarkino94, 2024-01-15:
> I see what you mean. It doesn't work the same between the 2 slicers.

Ok its possible to fix or to Add as orcaslicer ?

djngnear, 2024-01-15:
> > I see what you mean. It doesn't work the same between the 2 slicers.
> 
> Ok its possible to fix or to Add as orcaslicer ?

I would assume so. Hopefully they'll put it on their to-do list

closed 2024-05-16 by djmarkino94