bambulab/BambuStudio #3327: Purge into ______ checkmarks missing after newest update

she2dancer, 2024-01-07:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 12.6.4

### Printer

Bambu X-1C

### How to reproduce

Everytime I purge into an object the checkmarks are missing- as seen in pic
[image]

This is what it used to look like - 
[image]

### Actual results

[image]

### Expected results

The checkmarks should show up when selected (or disappear when unselected). Right now purging into an object is a blind guess until you slice. 

### Project file & Debug log uploads

I don't have this. 
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-01-08:
it been fixed , and you will get it in the next version.

LiZ-Li-BBL, 2024-01-16:
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.03.89
Get the latest version here.

closed 2024-01-16 by LiZ-Li-BBL