bambulab/BambuStudio #3328: Salmon Skin since Firmware update

HarrisonLDavies, 2024-01-07:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OSX Sonoma 14.2.1 (23C71)

### Additional system information

Apple M2 Pro, 16 GB

### Printer

Bambu Lab P1S

### How to reproduce

Print enclosed 3MF.

### Actual results

[image]
[image]
[image]
[image]

### Expected results

No Salmon skin.

I have cleaned all rods and tightened XY belt as per Wiki. I have recalibrated. This has only occurred with the new firmware install. Printer is a solid unmovable surface.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13855247/log.zip)

[sleeping dragon box.3mf.zip](https://github.com/bambulab/BambuStudio/files/13855242/sleeping.dragon.box.3mf.zip)

[RIGHT+ANGLE+METRIC-02.3mf.zip](https://github.com/bambulab/BambuStudio/files/13855249/RIGHT%2BANGLE%2BMETRIC-02.3mf.zip)

[Bambu Lab Scraper X1.3mf.zip](https://github.com/bambulab/BambuStudio/files/13855270/Bambu.Lab.Scraper.X1.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

TCC-up, 2024-01-08:
Hi, can you help us submit a ticket on handy to upload the device logs and let us know the ticket number? We need the device log to see the exact problem, thanks!

HarrisonLDavies, 2024-01-08:
Hi,

US240108350001

This is the ticket number. 

> Hi, can you help us submit a ticket on handy to upload the device logs and let us know the ticket number? We need the device log to see the exact problem, thanks!

US240108350001