bambulab/BambuStudio #3330: Unable to turn on advance settings in settings process

TwistedPrints3D, 2024-01-08:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

All Bambu printers

### How to reproduce

I can click on the advance settings tab

### Actual results

Nothing happens

### Expected results

Should show all advance settings

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13856307/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

QingZhangBambu, 2024-01-08:
have you opened the develop mode?

Choque, 2026-05-31:
Yea never mind, I was blind 😅 Didn't notice the option simply jumping to another place