bambulab/BambuStudio #3335: Trying to save a printer nozzle profile as user preset crashes the program.

MistressPlague, 2024-01-08:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. At the top left of Bambu Studio, choose a printer nozzle preset
2. Click the edit icon next to it
3. On the right in the popup, click the save icon, next to the search icon and advanced toggle.
4. In the save preset popup, change the name and try to save as user preset.

Context:
I was trying to save a 60 percent filament purge profile as user preset.

### Actual results

It crashes in a non-responding state. Even after 3 hours in wait testing, it never responded.
CPU usage is consistently within 2% of 20% usage on my hardware. Disk says 0 to 0.1 MB/s. Network shows from 0 to 11 Mbps.

### Expected results

It saving as a user preset, not crashing.

### Project file & Debug log uploads

[Context.zip](https://github.com/bambulab/BambuStudio/files/13866288/Context.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MistressPlague, 2024-01-08:
Looks to be stuck trying to do this in the log:
`process_task{ type:Backup, id:0, path:, object:0, extra:11} and interval = 10`

bambu123, 2024-01-09:
> Looks to be stuck trying to do this in the log: `process_task{ type:Backup, id:0, path:, object:0, extra:11} and interval = 10`
> 
> As you may see in the log, it hangs forever; effectively a crash. I left it open for ages.

Please ref this wiki and provider the DMP file to me.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

MistressPlague, 2024-01-11:
I'm not sure you realize that's a 1GB file..
How do you suppose I send a file that big?

bambu123, 2024-01-11:
> I'm not sure you realize that's a 1GB file.. How do you suppose I send a file that big?

Please use a cloud storage service and share the link with us.

MistressPlague, 2024-01-11:
It's uploading to onedrive now. And you're sure this dump wont contain any sensitive information; given it is a memory dump?

bambu123, 2024-01-11:
> It's uploading to onedrive now. And you're sure this dump wont contain any sensitive information; given it is a memory dump?

You may shared with email. Not here.
My email: chunmao.guo@bambulab.com

MistressPlague, 2024-01-11:
It has been sent.

bambu123, 2024-01-11:
> It has been sent.

It got stuck in an impossible place. Can you send another DMP file to further confirm the problem?

MistressPlague, 2024-01-11:
Really demotivating bug reporters by making them do what you should be doing on your end.
Doing this locally would not only reproduce the issue, but let you debug it, even making use of breakpointing.
How to reproduce this bug is in the original post. I tested it on multiple different systems, always happens saving A1 save filament from that project. I did not test other presets, but I doubt it will be any different.

I'm sending the second dump in email now. I wont send anything after this one, as frankly I'm sick of this back and forth.

bambu123, 2024-01-13:
Found that relate to the name of the current and new preset. If the new name is just before the old in alpha order. Then the saving will fall in a dead cycle. We will fix it asap. You can change the new name as a temporary solution.

> From: ***@***.***>
> Date:  Fri, Jan 12, 2024, 00:26
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> Really demotivating bug reporters by making them do what you should be doing on your end.
Doing this locally would not only reproduce the issue, but let you debug it, even making use of breakpointing.
How to reproduce this bug is in the original post. I tested it on multiple different systems, always happens saving A1 save filament from that project. I did not test other presets, but I doubt it will be any different.

closed 2024-01-23 by MistressPlague