bambulab/BambuStudio #3338: Measuring Tool

earth325, 2024-01-09:
Problem:

Inability to verify dimensions within Bambu Studio: Users currently lack a tool to directly measure model dimensions, leading to potential scaling issues and uncertainty.
Frustrating workarounds: Users are resorting to external software (e.g., Fusion 360) to check dimensions, hindering workflow efficiency.
Proposed Solution:

Implement a measurement tool within Bambu Studio: This tool would allow users to:
Measure distances between points, edges, and faces: Provide accurate measurements for various model elements.
Display measurements prominently: Ensure clear visibility of dimensions for easy verification.
Offer customizable units of measurement: Allow users to choose their preferred unit system (e.g., millimeters, inches).
Consider additional functionalities:
Measure angles: Calculate angles between model faces for greater versatility.
Calculate surface area and volume: Provide comprehensive dimensional information.

Benefits:

Enhanced accuracy and consistency: Ensure precise model scaling and print dimensions.
Improved workflow efficiency: Eliminate the need for external software, saving time and effort.
Increased user confidence: Provide reassurance that models are accurately sized for printing.
Streamlined design and preparation process: Facilitate smoother workflow within Bambu Studio.
Additional Context (optional):

Specific use cases where measurement is crucial: Scenarios where the tool would be particularly valuable (e.g., designing parts with tight tolerances, printing models that fit into specific assemblies).

xSepticSidx, 2024-01-13:
This tool is sorely missed from this program, almost every model I open I want to measure something so it fits.

walterwongbbl, 2024-01-18:
@xSepticSidx @earth325 
Your requirement is reasonable, and we will discuss the possibility of incorporating this feature.

earth325, 2024-02-26:
Is there any updates regarding this?