bambulab/BambuStudio #3344: Unable to use custom filaments on Offical Upgraded P1S

screech24, 2024-01-10:
### Bambu Studio Version

1.8.2

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS

### Printer

Upgraded p1p to p1s 

### How to reproduce

1. Create new custom filament for P1S
2. Using bambu handy app try to select that custom filament to print a model from makerworld.

### Actual results

Unable to select the newly created custom filament profile for P1S in the handy app when sending a print from makerworld.

### Expected results

You should be able to select the newly created custom filament profile.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

capsel22, 2024-01-10:
Same issue, this whole system needs reworked honestly.
Using latest 1.05.1 P1P to P1S printer and 1.8.3 Open Beta 2 Studio

1. Created custom filament
[image]
[image]

2. Custom filament is available in calibration screen
[image]

3. Issue #1. Run the calibration, entered the K factor. However "Manage Results" is missing
[image]

4. Issue #2. Custom filament not available in AMS selection
[image]

bam-winterfell, 2024-01-11:
hi@capsel22
Currently the P1 series does not support setting custom materials on AMS. We will fix this issue in the future.I am deeply sorry for any inconvenience caused to you

capsel22, 2024-01-11:
> hi@capsel22 @screech24 Currently the P1 series does not support setting custom materials on AMS. We will fix this issue in the future.I am deeply sorry for any inconvenience caused to you

ah ok, thanks. I swear I thought I read it was. Anyway.
The calibration still doesn't work, I did 2 calibrations yesterday, however the "manage results" button is missing, see my picture above.

xp-Yang, 2024-01-11:
@capsel22 
The P1 series printers only remember the latest K Factor value for each filament preset, so there is no "Manage Result" button for the P1 series. You should be able to check the current K Factor in Device page - AMS dialog.

capsel22, 2024-01-11:
> @capsel22 The P1 series printers only remember the latest K Factor value for each filament preset, so there is no "Manage Result" button for the P1 series. You should be able to check the current K Factor in Device page - AMS dialog.

ok thanks. Back to creating filament presets in Orca than. Studio needs some work, this is not user friendly design..

fritsjan, 2024-02-05:
with orca slicer I was able to set the custom filament type for the ams...
But when I switch back to bambulab studio, in the ams settings the filiment type is blank (and I cannot select custom filaments).
However, it does sync to the correct custom filaments in the prepare window...

screech24, 2024-02-05:
I am closing this issues as for some reason people can't seem to read english, and keep commenting things that are unrelated to this issue.

closed 2024-02-05 by screech24

bjw-s, 2024-08-21:
@screech24 Did you ever manage to get this working? I am seeing the same behavior on my upgraded P1P->P1S. Any custom filament that I create when using the P1S printer preset does not get synced to the AMS / Device screen. However, when I use the P1P as the base printer preset when creating the new custom filament it _does_ get synced to the AMS 🤔

screech24, 2024-08-21:
As far as i know nothing has changed with this. If you have a P1P you have
to use P1P filament profiles even if it has been upgraded to P1S. Wrong in
my opinion but doesnt seem they want to address this issue.

tsmith35, 2024-08-21:
> @screech24 Did you ever manage to get this working? I am seeing the same behavior on my upgraded P1P->P1S. Any custom filament that I create when using the P1S printer preset does not get synced to [...]

Quick question... Have you flipped the switch to indicate that the printer has been upgraded? It's accessible via the control panel and through Bambu Handy

[image]

bjw-s, 2024-08-21:
> Quick question... Have you flipped the switch to indicate that the printer has been upgraded? It's accessible via the control panel and through Bambu Handy
> 
> 
> 
> [image]
> 
> 

Thank you for thinking along. Unfortunately I have already set that in the app / on the printer, so I don't believe that to be the cause of the issue

tsmith35, 2024-08-21:
Ah, in that case, it sounds as if someone overlooked a detail when they introduced that switch. As I recall, it took a great deal of discussion before Bambu Lab relented and provided a switch.

screech24, 2024-08-21:
In my opinion that switch is pretty useless.. all it does is prevent an
error message popping up in the slicer when you slice something for a P1S
and send it to a P1P which you can still ignore and print anyway. Doesnt
help if you can't use P1S filament profiles on a p1p anyway LOL

bjw-s, 2024-08-23:
I've submitted a ticket and after some back and forth this is the response that I got from Bambu:

> Thank you for the reply. I have confirmed, indeed it is so. Because the Studio printer model is determined based on the SN, the printer shown here is still P1P. Therefore, for the filament presets for P1S, they cannot be selected here.
> However, even if you cannot select custom filament here, it will not affect the actual printing. You can choose generic PLA, and when sending the print, ensure you select the correct AMS slot.

So it looks like the only way is to create a dummy filament profile (based off Generic `$filament`) for a P1P printer so that the AMS learns about the filament, but then actually use the P1S profile. Not great, but at least there's a workaround.