bambulab/BambuStudio #3347: Coloring of neagtive parts

HorstBaumann, 2024-01-10:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

A1 mini

### How to reproduce

1. put an negative part in an object
2. color the object with the painting tool
3. slice it

### Actual results

result: the surrounding of the negative parts behave strange and keep the old filament/color
[image]

### Expected results

the surrounding of negative parts should have the same color, as the rest of the layer

### Project file & Debug log uploads

[test2.zip](https://github.com/bambulab/BambuStudio/files/13891195/test2.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-01-12:
Thanks for reporting this issue and we will consider how to improve it later.

zhimin-zeng-bambulab, 2024-01-15:
We currently export STL with only model information and no color information.

HorstBaumann, 2024-01-15:
no, it has nothing to do with colors. when i made a cut-out with a negative part and want to export the result as a STL, the cut-out is missing. the negative parts are not rendered on the export.
I just mention this, because I think, the processing of negative parts needs a little improvement.

zhimin-zeng-bambulab, 2024-01-16:
[image]
Could you try "Mesh boolean" in the right-click menu to see if it can meet your needs?

tinylabspace, 2025-03-20:
Is there a work around for the negative part coloring issue? I resorted to removing them and drilling holes in my model instead because I never managed to get rid of the incorrect white layers. My model is 2 colors, solid brown on the bottom, solid white on the top which should be 1 change. Instead, there are more than 60 when sliced and I've failed to find a way to prevent it other than removing the negative parts. I have it set to flush to infill, that's where the white grid is coming from.

[image]

tinylabspace, 2025-03-20:
And moments after I posted it yes there is. I changed the base color of the entire object to match the bottom layer color then painted the upper portion white. Now there is just 1 color change.

My expectation would be that walls around a negative part are printed in the color of the object it is part of, not the base color of the object.

aemravan, 2025-06-14:
are there any plans to correct this.. ? This is still an ongoing issue. It is particularly troublesome for me when doing negative thread cuts on valve stem caps that span multiple color layers. The negative parts always sliced to the base color of the model, and it if intersects any color region above the slicer will keep it the same color and introduce a color change for all of the impacted layers. It is very filament wasteful and time consuming.. 

We should have the ability to either manually paint negative parts before they are sliced, or have the ability to make the negative parts layer follow the outer layer color scheme.

MarkReneOtten, 2025-07-09:
Any new information? I have some projects there, the workaround is no an option. 
Do you think this is a minor bug? It isn't there is so much filamet poo cause of this bug. :-(

narze, 2026-02-07:
> The Mesh Boolean workaround did not work for my case. What did work is adding a normal modifier part around the negative part and change the filament color of the modifier to the desired color.

This works