bambulab/BambuStudio #3348: Skirt Glitch (8 loops on same layer-height)

NVNDO, 2024-01-10:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulabs P1S

### How to reproduce

Unfortunately I can not reproduce the scenario - but the setup is the following:

16 objects divided into 4 groups

Each group of 4 are assembled (in order to print them at once)

Skirt is set to 1 loop at 1st layer  (in order to prime the nozzle and avoiding lack of filament)

### Actual results

The skirt was printed 8 times on the same layer over and over again. 

https://www.youtube.com/watch?v=9UJct4ExCMo

### Expected results

Skirt prints only 1 time on first layer

### Project file & Debug log uploads

[16 Tags (group of 4) TEST.zip](https://github.com/bambulab/BambuStudio/files/13892995/16.Tags.group.of.4.TEST.zip)
[log_printed_at_19_32_cancelled_at_20_14.zip](https://github.com/bambulab/BambuStudio/files/13892996/log_printed_at_19_32_cancelled_at_20_14.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

NVNDO, 2024-01-11:
@QingZhangBambu 

The same error just happened again. 

Here are the files:
[Glitch FAIL 3mf.zip](https://github.com/bambulab/BambuStudio/files/13903478/Glitch.FAIL.3mf.zip)
[skirt_loop_error_log.zip](https://github.com/bambulab/BambuStudio/files/13903479/skirt_loop_error_log.zip)

It definitely has to do with "by object" and presumably with printing different colors. 

This time I managed to get it on camera: 

https://youtu.be/AQbs630xv28?si=jkES04eOQBO_3xO6

NVNDO, 2024-01-11:
I would like to add something else to this:

Sometimes the skirt does not cover each object (although "by object" is selected)
Instead the skirt goes around all objects - which is not correct. 

[image]
As you can see above there are multiple assemblies (which should count as individual objects)

[image]
And although "by object" is selected, the skirt does not cover each assembly - But the entire build-plate.

[image]
And sometimes its like it should be. Above you can seen the same scenario, different objects and a skirt on every object.

[image]
And here there are just 2 skirts.  ?!?!

TCC-up, 2024-01-12:
Hi, can you help raise a work order on the handy side and give us the work order number, we need to combine the printer logs together to see the issue. Thanks-

NVNDO, 2024-01-13:
@TCC-up Hey there, sorry but I don't know what you mean by "work order number" ? Where do I find that number...? Should I open a support ticket? I don't know what you mean...

NVNDO, 2024-04-19:
Hey there, sorry for not replying yet. I just didn't have the time to provoke the error again. I might do so next week to see if it still fails and open a ticket to provide more data. It's just very time-consuming.. 

With best regards