bambulab/BambuStudio #3350: Please add more placeholder for klipper, moonraker, mainsail

cromartic, 2024-01-10:
**Is your feature request related to a problem? Please describe.**
When I upload the gcode from Bambu Studio to Mainsail, then there are some metadata missing. You can see this in line 1 in the picture.

[image]

- Layer Height
- Nozzle Diameter
- Filament Name
- Filament Type
- Filament Usage
- Filament Weight
- Print Time
- Extruder Temp.
- Bed Temp.

Most of them are missing, because they are at the beginning of the gcode, but they are expected at the end.

Then i put following to the Machine end G-code, you can see the results in line 2 of the picture.

```
; nozzle_diameter = {nozzle_diameter[0]}
; filament_settings_id = {filament_settings_id[0]}
; filament_type = {filament_type[0]}
; estimated printing time (normal mode) = 
; first_layer_temperature = {first_layer_temperature[0]}
; first_layer_bed_temperature = {first_layer_bed_temperature[0]} 
```

But the following placeholder are not working:

```
; layer_height = {layer_height[0]}
; filament used [mm] = {filament used [mm][0]}
; total filament used [g] = {total filament used [g][0]}
```

In line 3 of the picture you can see the full metadata from Prusa Slicer.

**Describe the solution you'd like**

Please adopt these settings automatically when you change the G-Code type to Klipper.
Included:
```
; layer_height = {layer_height[0]}
; filament used [mm] = {filament used [mm][0]}
; total filament used [g] = {total filament used [g][0]}
```

**Describe alternatives you've considered**

Please add placeholders for:
```
; layer_height = {layer_height[0]}
; filament used [mm] = {filament used [mm][0]}
; total filament used [g] = {total filament used [g][0]}
```

**Additional context**
[image]

XunZhangBambu, 2024-01-11:
@cromartic  Hi , the issue addressed in this PR #3090  is similar to the one you raised.

cromartic, 2024-01-11:
@XunZhangBambu Hi, yes it is similar.