bambulab/BambuStudio #3351: Segmentation Fault when using AppImage via CLI on Ubuntu 22.04

dd-dkenney, 2024-01-11:
### Bambu Studio Version

1.8.02.06

### Where is the application from?

Bambu Lab github releases, Built from source by yourself or others

### OS version

Ubuntu 22.04

### Additional system information

DigitalOcean Droplet running Ubuntu 22.04 on x86_64

### Printer

BambuLab P1P 0.4mm Nozzle

### How to reproduce

Setup environment on Ubuntu 22.04 Server
Download AppImage from Github releases with wget
Attempt to slice STL with the following command:

```
/home/slicer/BambuStudio.AppImage --info --debug 6 --orient 1 --arrange 1 --allow-rotations --enable-support="0" --sparse-infill-density="20%" --slice="1-plate" --export-slicedata /home/slicer/server/output/tree-star --outputdir /home/slicer/server/output/tree-star  --load-settings "/Slicer/BambuStudio/resources/profiles/BBL/machine/Bambu Lab P1P 0.4 nozzle.json;/Slicer/BambuStudio/resources/profiles/BBL/process/0.20mm Standard @BBL P1P.json" --load-filaments "/Slicer/BambuStudio/resources/profiles/BBL/filament/PolyLite PLA @base.json" --export-3mf slice.3mf /home/slicer/server/uploads/tree-star.stl
```

### Actual results

Receive `Segmentation fault (core dumped)` response after successful slice.

### Expected results

File is successfully sliced and expected output files are provided.

### Project file & Debug log uploads

Happens with any STL - no project file produced due to segmentation fault.
[bambustudio_output.txt](https://github.com/bambulab/BambuStudio/files/13896017/bambustudio_output.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-01-11:
I don't believe gdb didn't provide information. At least we can see the crash stack.

dd-dkenney, 2024-01-11:
I can provide a coredump of the fault if you'd like - wasn't sure if it would be helpful since `bt` only gave me what was in the ticket above but have no problem handing one over if it helps!

bambu123, 2024-01-11:
I found no 'bt' output in the log above.

dd-dkenney, 2024-01-11:
[gdb_output.txt](https://github.com/bambulab/BambuStudio/files/13896563/gdb_output.txt)

Here is a run via `gdb` with `bt` output included

bambu123, 2024-01-11:
Ok, I will reproduce your operation.

dd-dkenney, 2024-01-11:
Investigating this a bit further, I realized something with Ubuntu 23.10 - the following packages were required to be installed before I could even run the AppImage which is somewhat confusing because AppImages are supposed to essentially be self contained - yet the BS AppImage seems to depend on libraries not included in the AppImage.

The reason I point this out is because I'm wondering if the segmentation fault is possibly related to a missing package that isn't being reported for some reason.

Here's a sample of the error I received from one of the run attempts:

```
root@ubuntu-g-4vcpu-16gb-nyc3-01:~# ./BambuStudio.AppImage --info --debug 6 --orient 1 --arrange 1 --allow-rotations --enable-support="0" --sparse-infill-density="20%" --slice="1-plate" --export-slicedata ~/ --outputdir ~/  --load-settings "/Slicer/BambuStudio/resources/profiles/BBL/machine/Bambu Lab P1P 0.4 nozzle.json;/Slicer/BambuStudio/resources/profiles/BBL/process/0.20mm Standard @BBL P1P.json" --load-filaments "/Slicer/BambuStudio/resources/profiles/BBL/filament/PolyLite PLA @base.json" --export-3mf slice.3mf ~/Tree_top_with_star-final.stl
/tmp/.mount_BambuSrpMdYc/bin/bambu-studio: error while loading shared libraries: libgstvideo-1.0.so.0: cannot open shared object file: No such file or directory
```

Upon installing the specified library, it would move to another until the following list of modules were installed, at which point it reached a segmentation fault with the exact same output as provided already for 22.04.

List of libraries that needed to be installed on a fresh Ubuntu 23.10:

```
apt install libgl1 libegl1 libgstreamer-plugins-base1.0-0 libwebkit2gtk-4.0-37
```

dd-dkenney, 2024-01-11:
Providing some more details hopefully, I created an EC2 instance with Ubuntu 22.04 LTS Desktop with RDP. I downloaded the AppImage from the repo (v1.8.3) and attempted to run the same command with the same results in Terminal with the same file.

However, I can open the AppImage and run BambuStudio GUI perfectly fine along with installing network plugin and slicing file.

bambu123, 2024-01-23:
> Providing some more details hopefully, I created an EC2 instance with Ubuntu 22.04 LTS Desktop with RDP. I downloaded the AppImage from the repo (v1.8.3) and attempted to run the same command with [...]

We are trying to to reproduce this issue.

dd-dkenney, 2024-02-02:
I just thought I'd share something additional here that may or may not help - I noticed I get similar results (segmentation fault) when using settings/parameters with other slicers via CLI that don't exist. For example, this happens with CuraEngine - if I pass a parameter which it doesn't know (i.e. a parameter added in a newer version than the one I am using in this instance) - it will not return an error but just encounter a Segmentation fault when slicing a file.

It made me think perhaps the same is happening here?

lanewei120, 2024-02-04:
please pass 0 to slice currently:
/home/slicer/BambuStudio.AppImage --info --debug 6 --orient 1 --arrange 1 --allow-rotations --enable-support="0" --sparse-infill-density="20%" **--slice="1-plate"** --export-slicedata /home/slicer/server/output/tree-star --outputdir /home/slicer/server/output/tree-star  --load-settings "/Slicer/BambuStudio/resources/profiles/BBL/machine/Bambu Lab P1P 0.4 nozzle.json;/Slicer/BambuStudio/resources/profiles/BBL/process/0.20mm Standard @BBL P1P.json" --load-filaments "/Slicer/BambuStudio/resources/profiles/BBL/filament/PolyLite PLA @base.json" --export-3mf slice.3mf /home/slicer/server/uploads/tree-star.stl

we should pass a number to --slice
and currently for stl slicing, we need to use 0 for slicing all

dd-dkenney, 2024-02-04:
Hey @lanewei120  - when I attempt to do this I receive the following error:

[2024-02-04 02:13:02.895261] [0x00007f65ed780c80] [info]    Need to slice for plate 0, total plate count 1 partplates!

[2024-02-04 02:13:02.895296] [0x00007f65ed780c80] [info]    Plate 1: pre_check 1, start
[2024-02-04 02:13:02.895311] [0x00007f65ed780c80] [debug]   BuildVolume printable_area clasified as: Rectangle
[2024-02-04 02:13:02.895753] [0x00007f65ed780c80] [info]    update_print_volume_state, print_volume {0, 0, 0} to {200, 200, 100}, got 0 printable istances
[2024-02-04 02:13:02.895765] [0x00007f65ed780c80] [error]   plate 1: Nothing to be sliced, Either the print is empty or no object is fully inside the print volume before apply.

I found a issue someone created surrounding this apparent bug - the gist of that issue was that when using `--arrange` it would always create an empty Plate 1 and arrange the item on Plate 0, leaving this scenario where the slicer complains that Plate 1 is empty even though plate 0 is specifically specified here via `--slice 0`

lanewei120, 2024-02-04:
slice 1 has a segment faut bug which we have just fixed internal, and will upload to github later
currently you can use slice 0 instead

and from your error
" plate 1: Nothing to be sliced, Either the print is empty or no object is fully inside the print volume before apply."
it is likely your stl not arranged onto plate 1, maybe too big?
you can try with another stl

there are more things need to take care in your command line ：
1. the config should be full config, 
[image]

dd-dkenney, 2024-02-04:
Now that you mention it, I do see:

update_print_volume_state, print_volume {0, 0, 0} to {200, 200, 100}, got 0 printable istances

Which is odd because I know the build volume should be larger than 200, 200, 100 - my model is slightly taller than 100mm so I can only assume based on your response then, that this _may_ be the cause of the issue I am having. Let me try something really quick and I will return shortly!

lanewei120, 2024-02-04:
that is just because you use the profile from resource directly which is not a full config

[image]

the p1p's size should be 256x256

[0.20mm Standard @BBL P1P.json](https://github.com/bambulab/BambuStudio/files/14155312/0.20mm.Standard.%40BBL.P1P.json)
[Bambu Lab P1P 0.4 nozzle.json](https://github.com/bambulab/BambuStudio/files/14155313/Bambu.Lab.P1P.0.4.nozzle.json)
[PolyLite PLA @BBL P1P.json](https://github.com/bambulab/BambuStudio/files/14155314/PolyLite.PLA.%40BBL.P1P.json)

I just attached the full files of P1P here

dd-dkenney, 2024-02-04:
That seemed to fix it - honestly, if the Segmentation Fault is already addressed when attempting to use `--slice 1` and `--slice 0` works - I think I might be good here - thank you so much for your help!

lanewei120, 2024-04-30:
already fixed in 1.9 public beta versions

closed 2024-04-30 by lanewei120