bambulab/BambuStudio #3352: After Printing PAHT-CF changing to PLA.  The nozzle doesn't purge the PAHT-CF because it only heats to 200 (not hot enough).  As a result the nozzle is partially clogged.

TimS-8, 2024-01-11:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I don't know it's a Surface Studio with an i7

### Printer

Bambu Lab X1C

### How to reproduce

With AMS print a project with PAHT-CF.  Then create a new project and try to print with PLA.  The nozzle will only heat to 200 when it purges (makes the poop).  As a result the PAHT-CF won't flow out.

My pseudo fix.... cancel job and then heat extruder to 290C.  Then manually tell it to extrude the PAHT-CF.  I had already removed the stub of PLA so this didn't completely work... but mostly

### Actual results

First it broke my PLA trying to load it and reported a nozzle jam.

After I semi-fixed the PAHT-CF jam by force extruding.  I attempted send the job again.  My green PLA was contaminated with PACF-HT (it was black).  I think the PAHT-CF isn't fully purged.

### Expected results

What should happen is that when the last job was PAHT-CF (or anything hotter than what we are printing), the next job should do the purge step using the previous filament temperature  OR the software should throw some sort of error and let the user know how to manually purge (be better if there was a software button to manually purge a higher temp filament)

### Project file & Debug log uploads

[tutk_log_iotc_22012.txt](https://github.com/bambulab/BambuStudio/files/13898096/tutk_log_iotc_22012.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

TimS-8, 2024-01-11:
NOTE!!!! the attached log file has nothing to do with it, but I couldn't submit the bug until I attached a log file... So ignore the log file.

pichen2021, 2024-01-22:
Hi @TimS-8,
    After printing PAHT-CF changing to PLA(assuming PLA basic), normally,  it will first purge with 250℃, and then purge while cooling cown to 220℃, and then wipe nozzle with 200℃, could you please share the printer log or the project file(3mf or gcode) for further analysis?
    After printing PAHT-CF, the inside of the nozzle may be partially clogged due to particle residue and accumulation, it is also recommended to refer to this [wiki ](https://wiki.bambulab.com/en/knowledge-sharing/common-print-quality-problem)to clean the nozzle.
[image]