bambulab/BambuStudio #3355: Can't increase raft-bed adhesion layer

Hologos, 2024-01-11:
### Bambu Studio Version

v1.8.3.89

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.2.1

### Printer

Bambu Lab P1S

### How to reproduce

1. insert an object
2. set raft layers (I set mine to 3)
3. enable supports
4. increase Raft->Initial layer expansion to 50mm + Advanced->Tree supports brim width to 50mm
5. No change

### Actual results

The contact layer does not expand. I have adhesion problems printing with BambuLab ASA and I need to increase the bed contact layer.

[image]

### Expected results

Increase in contact layer.

### Project file & Debug log uploads

Project file [DeLonghi EC850.m tray - tray v1.1.3mf.zip](https://github.com/bambulab/BambuStudio/files/13908756/DeLonghi.EC850.m.tray.-.tray.v1.1.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

denouche, 2024-03-08:
Hello
I'm facing the same issue. 
Any news on a fix?
Thanks

denouche, 2024-03-10:
Hello @Hologos
It seems that using this bug occurs when using tree supports.
When changing back to normal supports, the first layer expansion is well applied.
Do you notice the same behavior?
It could be a clue to help debugging...

bjkeeton, 2024-03-16:
This same thing is happening to me. It is happening regardless of the type of tree support I use.
[image]

closed 2024-06-12 by tangketan

Bathsheba, 2026-03-22:
This isn't fixed. I still see the same bug.