bambulab/BambuStudio #3357: Bambu Studio freezes and crashes when Trying to view Camera

Letts16, 2024-01-12:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home 

### Additional system information

Edition	Windows 10 Home
Version	22H2
Installed on	‎10/‎8/‎2020
OS build	19045.3930
Experience	Windows Feature Experience Pack 1000.19053.1000.0
Device name	LAPTOP-T50DN0PB
Processor	Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz   2.70 GHz
Installed RAM	8.00 GB (7.87 GB usable)
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

A1

### How to reproduce

While printing,
1 go to device
2 Click Play button on video screen 

### Actual results

Freeze, Not responding, have to stop application.  

### Expected results

To show the camera view.

### Project file & Debug log uploads

File too large to attach

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-01-12:
please refer:
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works