bambulab/BambuStudio #3359: Allow name change for the printed bed

Cobnut3D, 2024-01-12:
I often print multiple models one after the other, leaving my slicer open all day.  WIth other slicers, the exported GCode (or file) is named after the first object placed on the bed, but Bambu Studio appears to use the first object loaded for the entire time the program is open, regardless of what's on the bed.

Request:  Allow the name of the printed bed to be changed without restarting the program.

Sonicboom247, 2024-01-12:
i agree, this is quite annoying it does not take the name of the sliced file. Instead, it holds on to the name of the first object.  The only way i can get around this is to click add a plate, delete the old plate, and add the new object to the new plate.

Haidiye01, 2024-01-15:
Can you take screenshots or record a screen to illustrate your needs

Cobnut3D, 2024-01-15:
Thanks for the response.  The issue is with the naming of the model being
printed.  When you first open Bambu Studio the name is set to the name of
the first object printed and it remains that name even when that object is
deleted and other objects are loaded.  I've printed for an entire day and
by the late evening the prints are still named after an object I printed in
the morning.  There is no way to change this name, no way to set it
manually to make more sense.  Other slicers allow this.

This is really important when you're printing a lot of objects with similar
coded names and need to keep close track of what's being printed.

bstudio_1.png shows the "Prepare" tab.  Note that the name at the top is
"039_1d.stl" but the only object loaded is 039_1e.stl.

bstudio_2.png is the "Device" tab.  The name of the object being printed is
"039_1d", but it's actually 039_1e being printed.

bstudio_3.png is a screenshot from the Bambu App showing the
notifications.  Note that all the notifications have the same name, even
though they're clearly different objects.

Thanks.

Jon

Cobnut3D, 2024-01-15:
> I don't fully understand. Is there any picture to illustrate your
> problem.thanks

[image: bambu_1.jpg][image: bambu_2.jpg][image: bstudio_3.png]

Haidiye01, 2024-01-15:
Did your picture upload successfully？What I see is this，I didn't see anything
[image]

Haidiye00, 2024-05-14:
Please download and use the latest version. If there are still any issues, please use pictures or screenshots to describe your ideas. Thank you