bambulab/BambuStudio #3363: AUX Fan Override Option in Print Presets

ErnieBall, 2024-01-12:
**Is your feature request related to a problem? Please describe.**
When printing big parts, i find that the Aux fan on my Bambu Lab P1S often contributes to part lifting, while having little benefit otherwise (since on big parts there is enough time for layers to cool)
In Bambu Studio, the Aux Fan speed is defined in the filament profile, which generally makes sense. However I find myself wanting to turn the Aux fan off on some prints regardless of the used filament. 
Just like, for some prints you want support active or use a brim, while on other prints you dont.

**Describe the solution you'd like**
Having an override option for the Aux fan speed in the print preset (the presets where you define layer hight, infill etc.) would allow to easily turn of or fine tune the Aux fan speed on a print by print basis, making it a lot more convenient.

Personally i would be satisfied with a simple "Use AUX fan" - on/off toggle, but an even better implementation could
allow to define the layer at which the AUX fan turns on and the fan speed.

**Describe alternatives you've considered**
None

**Additional context**
None

Litcube, 2024-01-12:
Yes please. Aux fan has much less to do with filament and more to do with structure!

AKMonkey1, 2024-01-14:
It would be most useful to be able to turn the aux fan on at specific layers where bridging ir overhangs would benefit from the extra cooling, and off where it is not needed and just messes with the print surface (especially for parts near the back left corner).

AKMonkey1, 2024-01-14:
[image]

paddy75, 2024-01-27:
good idea. have the same problems with parts who are very close to the fan that they warp. now edited the settings for the filament but is saved as an additional filament right now

invisibleninja06, 2024-03-04:
When printing wall art or large flat prints in pla the aux fan has caused more harm than good. I would really like an easy to toggle setting to disable this when I feel it's needed.

Right now it's either edit the filament presets or disable the fan once it's done inspecting the first layer.