bambulab/BambuStudio #3365: Manual Flow Dynamics Calibration for PLA-CF on P1S not possible

umhh, 2024-01-12:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Ventura

### Printer

P1S

### How to reproduce

Start manual flow dynamics calibration for PLA-CF

### Actual results

Gives Info box "*Printing PLA-CF material with Stainless Steel may cause nozzle damage" but when accepting does not start calibration.

### Expected results

starting to print calibration pattern

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-01-15:
we will improve it in the future

umhh, 2024-01-15:
Nice tomhear.And in the meantime, how do I calibrate this Filament?Uwe

zhimin-zeng-bambulab, 2024-01-16:
[image]
You can set your nozzle type on the printer to hardened steel

tsmith35, 2024-01-18:
> I was looking for that setting, but could not find it anywhere in Bambu Studio. There is only a setting for the diameter of the nozzle, but not for the material.

You will need to change the setting from the printer control panel.
1. Go into the menu and highlight the _3rd icon from the top_.
2. Press the _right arrow_ once.
3. Press the _down arrow_ 6 times and **Accessories** should be highlighted.
4. Press the _right arrow_ once and **Nozzle** should be highlighted.
5. Press the _right arrow_ once more and you'll see a list of nozzles.
6. Choose the nozzle you want (0.4mm hardened steel) and press the _OK_ button.

You're all set!

zhimin-zeng-bambulab, 2024-01-18:
The P1S machine will not push nozzle information. This test will be deleted in the next version.

zhimin-zeng-bambulab, 2024-10-21:
It has been fixed in 1.10 Bera1.

closed 2024-10-21 by zhimin-zeng-bambulab