bambulab/BambuStudio #3366: Lock focus when dragging controls in Preview pane

n9yty, 2024-01-12:
**Is your feature request related to a problem? Please describe.**
When dragging the layer/step progress sliders in the Preview pane if the mouse moves off to the side of the control it starts to affect the view making the camera rotate/move/etc.

**Describe the solution you'd like**
Typically once you click and hold on a control the control retains control/focus even if your mouse moves outside of its bounds. It would be nice that as you drag up/down or left/right it would still just affect that setting even if your mouse pointer moves outside the bounds of the control

**Describe alternatives you've considered**
I have found no alternative except to be very carful as you slide the sliders.

xp-Yang, 2024-01-15:
The layer slider does hold the mouse focus if you are dragging. It should not happen that the camera rotate/move if the mouse moves outside the boundaries of the slider. 
So this issue is a bug that occurs in specific conditions. Is it a reproducible problem? Please provide information about your OS version, BambuStudio version, and additional necessary information.

LiZ-Li-BBL, 2024-01-15:
Hi,
The layer slider does hold the mouse focus if you are dragging. It should not happen that the camera rotate/move if the mouse moves outside the boundaries of the slider.
So this issue is a bug that occurs in specific conditions. Is it a reproducible problem? Please provide information about your OS version, BambuStudio version, and additional necessary information.

n9yty, 2024-01-15:
@LiZ-Li-BBL Thank you for the prompt response. I was just going to capture a video demonstrating it, when I decided to try a different mouse. Lo and behold, it seems my old trusty Logitech M570 trackball has micro-dropouts so it thinks I let go of the mouse button. I could not duplicate the problem with a different pointing device. So it is my problem, not yours. Thanks again!

closed 2024-01-15 by n9yty

n9yty, 2024-01-16:
I thought I had closed this. Strange. I double checked based on your information, and I was going to make a video demonstrating it because it was still happening to me, when I realized that my Logitech M570 trackball must have a dirty switch that lets up for tiny bits as I was dragging. I tried with another mouse and had no problems. Very odd, but my problem and not yours. Thanks!