bambulab/BambuStudio #3368: Proper profiles in Bambu Studio for Bambu Lab filaments

lindsayfowler, 2024-01-13:
**Is your feature request related to a problem? Please describe.**
I have just taken your PET G Basic out of the box and used the default settings for PETG Basic and printed 7 failed prints with it. What a waste of filament..
[image]

**Describe the solution you'd like**
I expect that when I select PETG in Bambu Studio, the correct settings will be applied throughout the project.
[image]

- For example, Layer height: 0.2mm - any less causes a mess. 
- All print speeds set to 180mm/s - that's Important.
[image]

[image]

**Describe alternatives you've considered**
The alternative is what I was forced to do - I had to spend days, dialling in my own settings, doing calibration to see if it was over extruding, trawling through the forums with hundreds of instances of. unofficial advice... Entering new settings into Bambulabs because the profile for PETG Basic didn't do it for me. And then there are all the recommendations to dry the filament for 8 f-ing hours!  None of this is ideal. So you need to put together an official guide to using this material and make Bambu Studio a proper profile for PETG Basic and for that matter any other filament you sell. 
[image]

**Additional context**
Add any other context or screenshots about the feature request here.
[image]
[image]
[image]
[image]

lindsayfowler, 2024-01-15:
> That last image looks like wet filament to me... perhaps you should dry it?

No, I really don't think so. Lots of people automatically assume the filament must be "wet" and spend 8 hours "drying" their filament in ovens. People even think their filament is "wet" when it is new in the bag. It's crazy.

The evidence:
1) First layer always goes down perfectly. That's interesting.. it seems to dig up the previous layer.
2) Calibration test prints perfectly. This is a big clue. How could a calibration test print perfectly if wet filament was causing the problem? What is different about the calibration test? The layer height is the major difference.

Setting the layer height to 0.2mm fixed the problem. This digging up of the previous layer is caused by the extra viscosity of PETG. 0.2mm layer height immediately stopped this problem, but the internal structure was not sticking.

To fix that I turned the speed down to 50% (Silent mode in Bambu Studio)

Model now prints perfectly.

And all these people are religiously cooking their filament in ovens for 8 or 9 hours unnecessarily.

FriedCheese2006, 2024-01-17:
The layer height isn't set in the filament profile. It's set under the print options which are separate from the filament profiles. Are you sure you didn't accidentally change the print options to 0.08 or modify the system preset? Using the 

The system default 0.2 print option has all layers set to .2mm. 
[image]

Using either of the 0.08mm presets will change the initial layer to .2 and all others to .08 like you're saying you are seeing:
[image]

lindsayfowler, 2024-01-24:
> Just got some PETG and did a print. Came out beautifully 🤷

No one said PETG doesn’t work.

JohnnyCarvi, 2024-02-12:
I am experiencing the same problems as @lindsayfowler with the Bambulab PETG Basic standard profile and Bambulab PETG Basic filament. It worked better with the Generic PETG, but still not perfectly. When I additionally activate silent mode (50% Speed), it comes out very well. Therefore, it would be good if Bambulab could readjust and test the PETG Basic profiles for their filament again.

JohnnyCarvi, 2024-02-12:
I have opened 2 brand new Bambulab PETG Basic filaments (Blue and Grey). Both have the same problem. They were also dried in an oven for about 8 hours. However, do you know the ambient temperature in our printer room? Mine is currently about 15 degrees Celsius. I’m not sure if this is causing the issue. Inside the printer about 25 degrees Celsius.

lindsayfowler, 2024-02-23:
> Just got some PETG and did a print. Came out beautifully 🤷

Well with the right settings it will print well. 🤷🏻‍♂️

lindsayfowler, 2024-02-23:
> I have opened 2 brand new Bambulab PETG Basic filaments (Blue and Grey). Both have the same problem. They were also dried in an oven for about 8 hours. However, do you know the ambient temperature [...]

I printed at 20 deg.C. Never needed to dry the filament. I think I’m going to store my filament underwater for a week and then do a YouTube video of me printing with it, just for the environmental impact it will have in showing people they don’t actually need to put their ovens on for eight hours every time they 3D print. 😂

mzient, 2024-07-02:
I also had problems with Bambu PETG profile. Things I did that helped:
- crank up nozzle temperature for other (than the first one) layers to 265*C
- crank up cooling:
    - 30% at 30s to 90% at 8s
    - 100% for overhangs
- run calibration (especially flow dynamics!)

The problems disappeared.

NOTE: I run a custom high-flow nozzle, so I guess that flow dynamics calibration might not be necessary with stock hotend. With that custom nozzle I was able to print up to 270mm/s and 22mm^3/s with no major issues.