bambulab/BambuStudio #3371: "Height to Rod" of 36mm causing collisions when using "By Object" print ordering.

Xaeco, 2024-01-14:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

OS Name: Microsoft Windows 11 Pro
Version: 10.0.22631 Build 22631
CPU: AMD Ryzen 9 7900 12-Core Processor, 3701 Mhz, 12 Core(s), 24 Logical Processor(s)
RAM: 64GB DDR5

### Printer

Bambu Lab P1P

### How to reproduce

1. Open Bambu Studio.
2. Select a P1P Printer.
3. Open Plate Settings, change the Print Sequence to By Object, click Ok.
4. In Process, select Global, select System Present "0.16mm Optimal @BBL P1P".
5. Add 2 cube primitives with a height of 35.85mm.
6. Place the cubes so they have the same Y position and are also far enough apart that there aren't any errors indicating they are too close. Make sure both cubes are still on the plate.
7. Print the plate.

### Actual results

Bambu Studio doesn't warn of height collisions.
Printing causes carbon rods to collide with first printed object.

### Expected results

Bambu Studio should error due to potential height collisions.

### Project file & Debug log uploads

[Example.zip](https://github.com/bambulab/BambuStudio/files/13930160/Example.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2024-01-22:
[image]
No issue on my X1C. Should be OK by P1P as well because they have same structure of rod.
Will find one P1P to have a try

Xaeco, 2024-01-22:
If "Height to rod" of 36mm is the exact physical distance from tip to rod then there isn't much tolerance for slight physical differences.

SaltWei, 2024-01-22:
> If "Height to rod" of 36mm is the exact physical distance from tip to rod then there isn't much tolerance for slight physical differences.

Yes. We can now reproduce this issue sometimes. This value should be smaller.
Thank you very much!

SaltWei, 2024-01-22:
> 6\. Place the cubes so they have the same Y position and are also far enough apart that there aren't any errors indicating they are too close. Make sure both cubes are still on the plate.

Same Y value is the important condition to reproduce this bug issue. We will reduce the setting to be smaller in next version.

@XunZhangBambu Please keep following.

XunZhangBambu, 2024-03-21:
@Xaeco  Thanks. We have set it to 34mm to avoid collision.

closed 2024-04-07 by XunZhangBambu