bambulab/BambuStudio #3373: Flatpak: Flushing Volumes Dialog not displayed correctly

kadrim, 2024-01-14:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 22.04

### Printer

Bambu Lab X1C

### How to reproduce

1. Install Bambu Studio from Flatpak
2. Start Bambu Studio
3. Switch to Tab "Prepare"
4. Add some Filaments
5. Click on "Flushing Volumes"

### Actual results

The Dialog is not displayed correctly:

[image]

### Expected results

Full Dialog Displayed, i.e.:

[image]

### Project file & Debug log uploads

no logs / project files needed to reproduce

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kadrim, 2024-01-14:
I first came across this issue on my steam deck (steam os / arch linux) and wanted to give the new flatpak version of bambu studio a try.

Afterwards i tried to find out if it was related to the flatpak build by using my ubuntu 22.04 machine as reference. On ubuntu the AppImage build is displaying the "flushing volumes" correctly. Then i installed the flatpak version also on my ubuntu22.04 and there the dialog is broken so most probably only related to the flatpak build.

as i am not sure if this should be a bug report in this repo, could @hadess also have look please?

kadrim, 2024-01-14:
if anything further is required, please let me know

hadess, 2024-01-14:
> as i am not sure if this should be a bug report in this repo, could @hadess also have look please?

Please report Flatpak specific bugs at https://github.com/flathub/com.bambulab.BambuStudio/issues

kadrim, 2024-01-14:
issue recreated here: https://github.com/flathub/com.bambulab.BambuStudio/issues/1

so closing this one

closed 2024-01-14 by kadrim