bambulab/BambuStudio #3377: Project Tabs for Bambu Studio

philsward, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
Bambu Studio will only let you work on one project at a time. You can open another instance of Bambu Studio and get the "network plugin could not install" error or just switch in and out of projects.

**Describe the solution you'd like**
Right now I'm in the middle of printing a project. I want to work on slicing, tweaking and testing a new project. Instead of opening a new instance, a new project in a new tab would be great, allowing multiple projects to be open at the same time.

**Describe alternatives you've considered**
The only workaround is a new instance of Bambu Studio.

**Additional context**
Mockup:
[image]

LJKraus, 2024-01-16:
Once a project is sent to the printer, you are free to do whatever you want in the Prepare/Preview tabs. You can start a new project, or open another project.  Once the sliced file is sent to the printer, nothing you do in Prepare/Preview will change the  print in progress.

hexxxx, 2024-01-18:
This would be an awesome addition!

SaltWei, 2024-01-18:
This is good suggestion. But not easy to be implemented at once.
Add into feature pool.

JonLaliberte, 2025-05-14:
Yeah, this would be great to have. Instead I end up with this mess 😬

[image]