bambulab/BambuStudio #3381: Calibraton>Tutorial - NOT AUTHORIZED TO VIEW TUTORIAL PAGE!

nebraskanerd, 2024-01-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Dell XPS
Processor: 13th Gen Intel(R) Core(TM) i9-13900H   2.60 GHz
RAM: 64 GB

### Printer

Bambu Labs X1 Carbon

### How to reproduce

1. Open Bambu Studio.
2. Click on Calibration in the upper left.
3. Then click tutorial.

### Actual results

You get a notification that you're not authorized to view tutorial.
[image]

### Expected results

Should get a page that shows you how to use the calibration feature in Bambu Studio.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LJKraus, 2024-01-15:
I don't see a Tutorial option in the calibration menu of Studio 1.8.2.56 nor the current 1.8.3.89 , though there is a wiki link on the top right.

OrcaSlicer has a separate calibration menu on the very top line which does have a [Tutorial link](https://github.com/SoftFever/OrcaSlicer/wiki/Calibration), but those procedures are different than the ones in Studio.

zhimin-zeng-bambulab, 2024-01-19:
If you are using a third-party printer preset for calibration (the calibration in the upper left corner), you can refer to OrcaSlicer's wiki (https://github.com/SoftFever/OrcaSlicer/wiki/Calibration). Our calibration process is consistent with theirs.

If you are using a Bambu printer for calibration, please refer to the wiki: 
https://wiki.bambulab.com/en/software/bambu-studio/calibration_pa
https://wiki.bambulab.com/en/software/bambu-studio/calibration_flow_rate

dragon-eyes-bbl, 2024-04-24:
This issue should be fixed and cannot reproduce in 1.8.4

closed 2024-04-24 by dragon-eyes-bbl