bambulab/BambuStudio #3382: Bambu Studio Crashing When Slicing Complex File

TheEightBitDad, 2024-01-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	13th Gen Intel(R) Core(TM) i9-13900KF   3.00 GHz
Installed RAM	32.0 GB (31.8 GB usable)
Video Card RTX 4080
System type	64-bit operating system, x64-based processor

### Printer

X1/P1S/A1

### How to reproduce

Cllick "slice" on complex or large model

### Actual results

Bambu Studio Crash

### Expected results

Not Crashing

### Project file & Debug log uploads

Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0xfa8542d7 
Context :000000E4DE0F51C0 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0xF297CDD3 0x0:0x0 

Show CallStack:
Call SymGetLineFromAddr64 ,Address f297cdd3 Error:000001e7
Call SymGetLineFromAddr64 ,Address f29797d9 Error:000001e7
Call SymGetLineFromAddr64 ,Address f2980e07 Error:000001e7
Call SymGetLineFromAddr64 ,Address f297bf57 Error:000001e7
Call SymGetLineFromAddr64 ,Address f297b5f7 Error:000001e7
Call SymGetLineFromAddr64 ,Address f297b773 Error:000001e7
Call SymGetLineFromAddr64 ,Address f29aa5df Error:000001e7
Call SymGetLineFromAddr64 ,Address f299cb2b Error:000001e7
Call SymGetLineFromAddr64 ,Address f299c453 Error:000001e7
Call SymGetLineFromAddr64 ,Address f29a9720 Error:000001e7
Call SymGetLineFromAddr64 ,Address f2ca9b93 Error:000001e7
Call SymGetLineFromAddr64 ,Address f2b9e0d5 Error:000001e7
Call SymGetLineFromAddr64 ,Address f2afdfb1 Error:000001e7
Call SymGetLineFromAddr64 ,Address f2acb598 Error:000001e7
Call SymGetLineFromAddr64 ,Address f2aa9e78 Error:000001e7
Call SymGetLineFromAddr64 ,Address f2ac3649 Error:000001e7
Call SymGetSymFromAddr64 ,Address f2623960 Error:000001e7
Call SymGetSymFromAddr64 ,Address f261ef82 Error:000001e7
Call SymGetSymFromAddr64 ,Address f262a2dc Error:000001e7
Call SymGetSymFromAddr64 ,Address f26317a7 Error:000001e7
Call SymGetSymFromAddr64 ,Address f263166b Error:000001e7
Call SymGetLineFromAddr64 ,Address 62d89363 Error:000001e7
Call SymGetLineFromAddr64 ,Address 644e257d Error:000001e7
Call SymGetLineFromAddr64 ,Address 6536aa58 Error:000001e7

Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x2b0f5690 
Context :000000C32B0F4A80 
ContextFlag : 0x10005f, EFlags: 0x10202 
Fault address:  0x2FFD4B 0x0:0x0 

Show CallStack:
Call SymGetLineFromAddr64 ,Address 002ffd4b Error:000001e7
Call SymGetLineFromAddr64 ,Address 002fa425 Error:000001e7
Call SymGetLineFromAddr64 ,Address 002fb72a Error:000001e7
Call SymGetLineFromAddr64 ,Address 0032a5df Error:000001e7
Call SymGetLineFromAddr64 ,Address 0031cb2b Error:000001e7
Call SymGetLineFromAddr64 ,Address 0031c453 Error:000001e7
Call SymGetLineFromAddr64 ,Address 00329720 Error:000001e7
Call SymGetLineFromAddr64 ,Address 00625eda Error:000001e7
Call SymGetLineFromAddr64 ,Address 00628e8d Error:000001e7
Call SymGetLineFromAddr64 ,Address 0051e0d5 Error:000001e7
Call SymGetLineFromAddr64 ,Address 0047dfb1 Error:000001e7
Call SymGetLineFromAddr64 ,Address 0044b598 Error:000001e7
Call SymGetLineFromAddr64 ,Address 00429e78 Error:000001e7
Call SymGetLineFromAddr64 ,Address 00443649 Error:000001e7
Call SymGetSymFromAddr64 ,Address fffa3960 Error:000001e7
Call SymGetSymFromAddr64 ,Address fff9ef82 Error:000001e7
Call SymGetSymFromAddr64 ,Address fffaa2dc Error:000001e7
Call SymGetSymFromAddr64 ,Address fffb17a7 Error:000001e7
Call SymGetSymFromAddr64 ,Address fffb166b Error:000001e7
Call SymGetLineFromAddr64 ,Address 62d89363 Error:000001e7
Call SymGetLineFromAddr64 ,Address 644e257d Error:000001e7
Call SymGetLineFromAddr64 ,Address 6536aa58 Error:000001e7

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2024-01-22:
@QingZhangBambu 哪里能看出来是clipper问题，符号和调用栈贴上来

QingZhangBambu, 2024-01-25:
@TheEightBitDad Could you provide the 3mf and dump file?
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

QingZhangBambu, 2024-01-26:
@TheEightBitDad there are a method that some people think is useful, may you try it?https://github.com/bambulab/BambuStudio/issues/3355

closed 2024-08-06 by SaltWei