bambulab/BambuStudio #3383: MicroSD Card page in Device tab not showing new timelapses without reinstalling MicroSD Card

skorpeyon, 2024-01-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel Core i9-14900K, RAM: 64GB 4800 MHz DDR5, GPU: NVIDIA GeForce RTX 4090.

### Printer

Bambu Lab P1S

### How to reproduce

Print an item, set it to record timelapse. Wait for item to complete printing. Go to Device tab, click on "MicroSD Card" on the left. Click on "All Files" if not already selected.

Go to the printer. Select option to eject MicroSD Card. Remove and reinsert MicroSD Card. Return to Device Tab in Bambu Studio, select a different tab, then return to the Device tab to refresh the page. Click on MicroSD Card if it isn't already selected. Timelapses now show up.

### Actual results

No new timelapse is showing on Device tab, MicroSD Card, even though one has been recorded and is stored on the machine's MicroSD Card.

### Expected results

New timelapse should show up on this page after being recorded, should not require the card to be removed and reinserted.

### Project file & Debug log uploads

[Dragon_v2_MMU_Bambu.zip](https://github.com/bambulab/BambuStudio/files/13942106/Dragon_v2_MMU_Bambu.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bambu123, 2024-01-16:
Yes, a refresh button is missing. If you leave the SD card page for a while and come back, it should refresh as well.

skorpeyon, 2024-01-16:
I have left it for days, closing out of the program entirely, and not seen
a refresh button. I just did this today. I'd printed several things over
the past few days, there was no refresh button and none of them were
showing up.

bambu123, 2024-01-16:
> I have left it for days, closing out of the program entirely, and not seen a refresh button. I just did this today. I'd printed several things over the past few days, there was no refresh button and none of them were showing up.
> […](#)

What I mean is that a refresh button should be added here. The current version does not have it, and the new version will consider it.

skorpeyon, 2024-01-16:
I meant that I've also left it for days and it hasn't refreshed on its own.
It seems to never do so unless I remove and reinsert the Micro SD Card. I
agree that a refresh button would be a reasonable solution here, hope it's
able to be added.