bambulab/BambuStudio #3384: Support Enhancer Does Not Render Support Structures

T3chC0wb0y, 2024-01-16:
### Bambu Studio Version

1.8.3.89

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sonoma 14.2.1

### Additional system information

Apple M2 Pro
32 GB RAM

### Printer

Bambu X1C

### How to reproduce

Add a support enhancer
On Global Settings, enable Support > Normal (manual)
Click Slice Plate

### Actual results

No supports are rendered.

### Expected results

Custom supports should be rendered with the predefined gap established in the support settings.

### Project file & Debug log uploads

[Support_Enhancer_Broken.3mf.zip](https://github.com/bambulab/BambuStudio/files/13945654/Support_Enhancer_Broken.3mf.zip)
[Logs.zip](https://github.com/bambulab/BambuStudio/files/13945679/Logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

T3chC0wb0y, 2024-01-16:
[image]
[image]
[image]

T3chC0wb0y, 2024-01-16:
This is also referenced in ticket #420 but somehow it was resolved. I deleted the app and reinstalled and it did not resolve the issue.

tangketan, 2024-02-04:
This is not how support enforcer is supposed to be used. Support enforcer means: "the overhangs inside the enforcer volume will be supported". In your case, your enforcer didn't enclose any overhangs.

closed 2026-11-03 by ArthurBambulab