bambulab/BambuStudio #3385: 拓竹能否给出手动退料/进料的GCODE

LittleOAndLittleQ, 2024-01-16:
拓竹能否给出在打印机屏幕或studio、handy上点击退料/进料的GCODE？
我要的不是AMS进料退料的GCODE。感谢！
[image]
[image]

walterwongbbl, 2024-01-16:
@LittleOAndLittleQ 
https://github.com/bambulab/BambuStudio/blob/master/src/slic3r/GUI/StatusPanel.cpp
代码在on_ams_load_curr的相关实现中

GCODE代码看文件
[image]

LittleOAndLittleQ, 2024-01-16:
您好，我看了您给出的这两段代码，ams_unload代码中似乎没有切刀切断耗材和控制挤出电机的部分，这段代码是否有缺失？

 

 https://github.com/bambulab/BambuStudio/blob/master/src/slic3r/GUI/StatusPanel.cpp
 
 image.png (view on web)

walterwongbbl, 2024-01-18:
@LittleOAndLittleQ 一部分操作不是由Studio发送G代码完成的，而是打印机的行为。

closed 2024-02-04 by walterwongbbl