bambulab/BambuStudio #3387: BambuStudio forgets last used printer after restarting

vordenken, 2024-01-16:
### Bambu Studio Version

1.8.3.89

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sonoma

### Additional system information

MacBook Air M2

### Printer

BambuLab P1S

### How to reproduce

1. Use printer in lan-mode
2. select printer in bambuStudio "Device" tab
3. restart BambuStudio

### Actual results

BambuStudio has no printer selected after the restart.

### Expected results

It should remember the last selected printer.

### Project file & Debug log uploads

[tutk_log_avapi_49019.txt](https://github.com/bambulab/BambuStudio/files/13950058/tutk_log_avapi_49019.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2024-01-18:
Hi. @vordenken 
this is issue. We should remember LAN mode for the printer.

vordenken, 2024-10-22:
?

vordenken, 2024-12-06:
This is still an ongoing issue (at least for me). BambuStudio "forgets" the printer after a bit of inactivity or after my Mac "slept". Sometimes it even completely disappears so I have to restart BambuStudio to be able to find it again (This is even more frustrating when I have stuff with custom settings open).
Please make this a priority!

(I'm on the latest BambuStudio release now).

hayden-servaenergy, 2025-05-06:
Still an issue in Bambu Studio 2.0.3.54. Pretty annoying bug to deal with when printing in LAN-only mode. What is the timeline to implement this?

elexwiz, 2025-09-28:
Still an issue today in Bambu Studio 2.2.2.56. Printing in LAN-only mode I have to recapture the printer using the Access Code every time I restart the app. Is a fix on the way?