bambulab/BambuStudio #3392: Calibration Flow UI incorrect

designgears, 2024-01-16:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Go to Device tab
2. Click on Calibration button

### Actual results

Incorrect step is show in the app and on the printer display.

`Sweeping XY mech mode` is highlighted when doing `Motor noise calibration`
`Motor noise calibration` is highlighted when doing `Sweeping XY mech mode`

[image]
[image]

### Expected results

The correct step should be highlighted for what is physically happening on the printer.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13955305/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bam-winterfell, 2024-01-22:
hi@designgears 
Thank you for raising the question. I will try to reproduce this issue

walterwongbbl, 2024-05-21:
Please use our latest 1.9.1 version

closed 2024-05-21 by walterwongbbl